ros-perception/perception_pcl #350: Need PCL_ROS, when do catkin build

kennydedavion, 2021-11-07:
Good day.  can someone please explain to me why this repo cannot be built?  and what other PCL_ROS repo allows to build?  i need it all for the JSK package.  but "sudo apt-get install ros-melodik-pcl-ros" does not work.

thschafer, 2021-11-10:
I'm looking to figure out how to build this myself, but perhaps you can install with "sudo apt install ros-melodic-pcl-ros". Note the spelling of "melodic".

thschafer, 2021-11-10:
You can pull this repository into the src/ directory of a catkin_workspace and build it using catkin_make

kennydedavion, 2021-11-11:
Many thanks for the quick reply, and sorry "melodik" is the job of translation. Of course I tried to use "sudo apt-get install ros-melodic-pcl-ros" and the variant without "-get", and raspian returned that the package could not be found. You don't have to be sad about that. Many things I would like to use cannot be installed this way. I would extend the original question with an additional question. "jsk_visualization" I just wanted to "pie hart" display the temperature in rviz, like a bar graph. After I downloaded jsk visualization to SRC, I reached the missing dependency "jsk_common" with catkim_make. with catkin_make jsk_common, there was an error that the package "pcl_ros" was missing and I found only one on the github and it can't give catkin_make or catkin build.