ros-perception/perception_pcl #355: Unnecessary dependencies on Desktop packages in Ubuntu 20.04 + Noetic

civerachb-cpr, 2022-01-31:
When building or installing this package on Ubuntu 20.04 with a command-line only installation, the following additional dependencies get installed:

```
$ sudo apt-get install ros-noetic-pcl-ros
[sudo] password for administrator: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  acl apg aptdaemon aptdaemon-data aspell aspell-en avahi-daemon avahi-utils bubblewrap cheese-common colord colord-data cracklib-runtime cups-pk-helper cython3 dbus-x11 dconf-cli desktop-file-utils dictionaries-common
  dns-root-data dnsmasq-base docbook-xml emacsen-common enchant-2 evolution-data-server evolution-data-server-common fprintd freeglut3 gcr gdm3 geoclue-2.0 gir1.2-accountsservice-1.0 gir1.2-atk-1.0 gir1.2-atspi-2.0
  gir1.2-freedesktop gir1.2-gck-1 gir1.2-gcr-3 gir1.2-gdesktopenums-3.0 gir1.2-gdkpixbuf-2.0 gir1.2-gdm-1.0 gir1.2-geoclue-2.0 gir1.2-gnomebluetooth-1.0 gir1.2-gnomedesktop-3.0 gir1.2-graphene-1.0 gir1.2-gtk-3.0
  gir1.2-gweather-3.0 gir1.2-ibus-1.0 gir1.2-json-1.0 gir1.2-mutter-6 gir1.2-nm-1.0 gir1.2-nma-1.0 gir1.2-notify-0.7 gir1.2-pango-1.0 gir1.2-polkit-1.0 gir1.2-rsvg-2.0 gir1.2-secret-1 gir1.2-soup-2.4 gir1.2-upowerglib-1.0
  gir1.2-vte-2.91 gjs gkbd-capplet gnome-control-center gnome-control-center-data gnome-control-center-faces gnome-desktop3-data gnome-keyring gnome-keyring-pkcs11 gnome-menus gnome-online-accounts gnome-session-bin
  gnome-session-common gnome-settings-daemon gnome-settings-daemon-common gnome-shell gnome-shell-common gnome-startup-applications gnome-user-docs gstreamer1.0-clutter-3.0 gstreamer1.0-gl gstreamer1.0-plugins-good
  gstreamer1.0-pulseaudio gstreamer1.0-x hdf5-helpers hunspell-en-us ibus ibus-data ibus-gtk ibus-gtk3 iio-sensor-proxy im-config ippusbxd language-selector-gnome libaa1 libaec-dev libappindicator3-1 libarmadillo-dev
  libarpack2-dev libasound2-plugins libaspell15 libavahi-core7 libavahi-glib1 libavc1394-0 libblas-dev libbluetooth3 libcaca0 libcamel-1.2-62 libcanberra-gtk3-0 libcanberra-gtk3-module libcanberra-pulse libcfitsio-dev
  libcfitsio-doc libcharls-dev libcheese-gtk25 libcheese8 libclang1-10 libclutter-1.0-0 libclutter-1.0-common libclutter-gst-3.0-0 libclutter-gtk-1.0-0 libcogl-common libcogl-pango20 libcogl-path20 libcogl20 libcolord-gtk1
  libcolorhug2 libcrack2 libcurl4-gnutls-dev libdaemon0 libdap-dev libdapserver7v5 libdbusmenu-glib4 libdbusmenu-gtk3-4 libdouble-conversion3 libdv4 libebackend-1.2-10 libebook-1.2-20 libebook-contacts-1.2-3 libecal-2.0-1
  libedata-book-1.2-26 libedata-cal-2.0-1 libedataserver-1.2-24 libedataserverui-1.2-2 libegl-dev libegl-mesa0 libegl1 libenchant-2-2 libepsilon-dev libflann-dev libflann1.9 libfontconfig1-dev libfontenc1 libfprint-2-2
  libfreetype-dev libfreetype6-dev libfreexl-dev libfyba-dev libgail-common libgail18 libgbm1 libgck-1-0 libgcr-base-3-1 libgcr-ui-3-1 libgdal-dev libgdata-common libgdata22 libgdm1 libgee-0.8-2 libgeoclue-2-0 libgeocode-glib0
  libgeos-dev libgeotiff-dev libgif-dev libgjs0g libgl-dev libgl1-mesa-dev libgl2ps-dev libgles-dev libgles1 libgles2 libglew-dev libglew2.1 libglu1-mesa libglu1-mesa-dev libglvnd-dev libglx-dev libgnome-autoar-0-0
  libgnome-bluetooth13 libgnome-desktop-3-19 libgnomekbd-common libgnomekbd8 libgoa-1.0-0b libgoa-1.0-common libgoa-backend-1.0-1 libgraphene-1.0-0 libgsound0 libgssdp-1.2-0 libgstreamer-gl1.0-0 libgstreamer-plugins-good1.0-0
  libgtk2.0-0 libgtk2.0-bin libgtk2.0-common libgtop-2.0-11 libgtop2-common libgupnp-1.2-0 libgupnp-av-1.0-2 libgupnp-dlna-2.0-3 libgweather-3-16 libgweather-common libharfbuzz-icu0 libhdf4-alt-dev libhdf5-cpp-103 libhdf5-dev
  libhdf5-mpi-dev libhdf5-openmpi-dev libhunspell-1.7-0 libhyphen0 libibus-1.0-5 libice-dev libidn11 libiec61883-0 libieee1284-3 libimobiledevice6 libinput-bin libinput10 libjack-jackd2-0 libjansson4 libjavascriptcoregtk-4.0-18
  libjson-c-dev libjsoncpp-dev libkml-dev libkmlconvenience1 libkmlregionator1 libkmlxsd1 liblapack-dev libldb2 libllvm10 libmbim-glib4 libmbim-proxy libmediaart-2.0-0 libminizip-dev libmm-glib0 libmozjs-68-0 libmtdev1
  libmutter-6-0 libndp0 libnetcdf-cxx-legacy-dev libnetcdf-dev libnm0 libnma0 libnotify4 libnss-mdns libogdi-dev libogg-dev libopengl-dev libopengl0 libopenjp2-7-dev libopenni-dev libopenni-sensor-pointclouds0 libopenni0
  libopenni2-0 libopenni2-dev libpam-fprintd libpam-gnome-keyring libpangoxft-1.0-0 libpcl-apps1.10 libpcl-common1.10 libpcl-dev libpcl-features1.10 libpcl-filters1.10 libpcl-io1.10 libpcl-kdtree1.10 libpcl-keypoints1.10
  libpcl-ml1.10 libpcl-octree1.10 libpcl-outofcore1.10 libpcl-people1.10 libpcl-recognition1.10 libpcl-registration1.10 libpcl-sample-consensus1.10 libpcl-search1.10 libpcl-segmentation1.10 libpcl-stereo1.10 libpcl-surface1.10
  libpcl-tracking1.10 libpcl-visualization1.10 libpcre2-16-0 libphonenumber7 libplist3 libpoppler-dev libpoppler-private-dev libpq-dev libproj-dev libpthread-stubs0-dev libpulse-mainloop-glib0 libpulsedsp libpwquality-common
  libpwquality1 libqhull-dev libqhull-r7 libqmi-glib5 libqmi-proxy libqt5concurrent5 libqt5core5a libqt5dbus5 libqt5designer5 libqt5designercomponents5 libqt5gui5 libqt5help5 libqt5network5 libqt5opengl5 libqt5opengl5-dev
  libqt5positioning5 libqt5printsupport5 libqt5qml5 libqt5quick5 libqt5quickparticles5 libqt5quickshapes5 libqt5quicktest5 libqt5quickwidgets5 libqt5sensors5 libqt5sql5 libqt5sql5-sqlite libqt5svg5 libqt5test5 libqt5webchannel5
  libqt5webkit5 libqt5webkit5-dev libqt5widgets5 libqt5x11extras5 libqt5xml5 librygel-core-2.6-2 librygel-db-2.6-2 librygel-renderer-2.6-2 librygel-server-2.6-2 libsamplerate0 libsane libsane-common libsbc1 libsecret-1-0
  libsecret-common libshout3 libsm-dev libsmbclient libsnapd-glib1 libsnmp-base libsnmp35 libspatialite-dev libspeexdsp1 libstartup-notification0 libsuperlu-dev libtag1v5 libtag1v5-vanilla libtalloc2 libteamdctl0 libtevent0
  libtheora-dev libtk8.6 libupower-glib3 liburiparser-dev libusbmuxd6 libv4l-0 libv4lconvert0 libvte-2.91-0 libvte-2.91-common libvtk7-dev libvtk7-java libvtk7-jni libvtk7-qt-dev libvtk7.1p libvtk7.1p-qt libvulkan-dev
  libwacom-bin libwacom-common libwacom2 libwayland-server0 libwbclient0 libwebkit2gtk-4.0-37 libwebp-dev libwebrtc-audio-processing1 libwhoopsie-preferences0 libwhoopsie0 libwoff1 libx11-dev libxatracker2 libxau-dev
  libxcb-icccm4 libxcb-image0 libxcb-keysyms1 libxcb-render-util0 libxcb-res0 libxcb-shape0 libxcb-util1 libxcb-xinerama0 libxcb-xinput0 libxcb-xkb1 libxcb-xv0 libxcb1-dev libxdmcp-dev libxerces-c-dev libxext-dev libxfont2
  libxft-dev libxft2 libxkbcommon-x11-0 libxkbfile1 libxklavier16 libxml2-dev libxrender-dev libxss-dev libxt-dev libxv1 libxvmc1 libxxf86dga1 libyelp0 libzstd-dev mobile-broadband-provider-info modemmanager mousetweaks mutter
  mutter-common network-manager network-manager-gnome network-manager-pptp openni-utils p11-kit p11-kit-modules pinentry-gnome3 ppp pptp-linux pulseaudio pulseaudio-module-bluetooth pulseaudio-utils python3-aptdaemon
  python3-aptdaemon.gtk3widgets python3-autobahn python3-cairo python3-cbor python3-cups python3-cupshelpers python3-defer python3-ibus-1.0 python3-ldb python3-lz4 python3-macaroonbakery python3-mpi4py python3-png
  python3-protobuf python3-pyqrcode python3-rfc3339 python3-snappy python3-talloc python3-trie python3-txaio python3-tz python3-u-msgpack python3-ubjson python3-vtk7 python3-wsaccel qdoc-qt5 qhelpgenerator-qt5 qt5-assistant
  qt5-gtk-platformtheme qt5-qmake qt5-qmake-bin qt5-qmltooling-plugins qtattributionsscanner-qt5 qtbase5-dev qtbase5-dev-tools qtchooser qtdeclarative5-dev qtdeclarative5-dev-tools qttools5-dev qttools5-dev-tools
  qttools5-private-dev qttranslations5-l10n ros-noetic-pcl-conversions ros-noetic-pcl-msgs ros-noetic-tf2-eigen rtkit rygel samba-libs sane-utils session-migration sgml-data switcheroo-control system-config-printer
  system-config-printer-common system-config-printer-udev tcl tcl-dev tcl-vtk7 tcl8.6-dev tk tk-dev tk8.6 tk8.6-dev ubuntu-docs ubuntu-session ubuntu-wallpapers ubuntu-wallpapers-focal unixodbc-dev unzip update-inetd upower
  usb-modeswitch usb-modeswitch-data usbmuxd vtk7 wamerican whoopsie-preferences x11-utils x11-xkb-utils x11-xserver-utils x11proto-core-dev x11proto-dev x11proto-scrnsaver-dev x11proto-xext-dev xbitmaps xdg-dbus-proxy
[... cut ...] itional steps are taken to reconfigure the unwanted desktop session.

Packages like `cheese` and `aspell` are unnecessary additions that eat up disk space, but aren't otherwise harmful.  But I'm confused about why they're required as dependencies for a point-cloud processing package.

Has anyone else used `pcl_ros` in a command-line-only Ubuntu environment and seen this issue?  And if you have, do you have a work-around to prevent the desktop packages and their dependencies from being installed?

mvieth, 2022-01-31:
Did you try to find out which package is required through which other package, for example with the answers from [this stackoverflow question](https://askubuntu.com/questions/132872/how-can-i-determine-why-apt-get-will-install-a-package)?

civerachb-cpr, 2022-01-31:
See the attached output.  The sheer number of dependencies is a little overwhelming.  A few stand-out packages that make me scratch my head are things like `abiword` and the various `aspell-*` dictionaries.

I _think_ what's happening is that apt is installing recommended packages by default, and something in the dependencies has changed between 18.04 and 20.04 to cause a lot of unnecessary packages to be brought in because of that.

Installing the package with
```
sudo apt-get install ros-noetic-pcl-ros --no-install-recommends
```
appears to mitigate the problem somewhat:
```
$ sudo apt-get install ros-noetic-pcl-ros --no-install-recommends

Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  cython3 hdf5-helpers libaec-dev libarmadillo-dev libarpack2-dev libblas-dev libcfitsio-dev libcharls-dev libclang1-10 libcurl4-gnutls-dev libdap-dev libdapserver7v5 libdouble-conversion3 libegl-dev libegl-mesa0 libegl1
  libepsilon-dev libflann-dev libflann1.9 libfontconfig1-dev libfreetype-dev libfreetype6-dev libfreexl-dev libfyba-dev libgbm1 libgdal-dev libgeos-dev libgeotiff-dev libgif-dev libgl-dev libgl1-mesa-dev libgl2ps-dev
  libgles-dev libgles1 libgles2 libglew-dev libglew2.1 libglu1-mesa libglu1-mesa-dev libglvnd-dev libglx-dev libhdf4-alt-dev libhdf5-cpp-103 libhdf5-dev libhdf5-mpi-dev libhdf5-openmpi-dev libhyphen0 libice-dev libinput-bin
  libinput10 libjson-c-dev libjsoncpp-dev libkml-dev libkmlconvenience1 libkmlregionator1 libkmlxsd1 liblapack-dev libllvm10 libminizip-dev libmtdev1 libnetcdf-cxx-legacy-dev libnetcdf-dev libogdi-dev libogg-dev libopengl-dev
  libopengl0 libopenjp2-7-dev libopenni-dev libopenni0 libopenni2-0 libopenni2-dev libpcl-apps1.10 libpcl-common1.10 libpcl-dev libpcl-features1.10 libpcl-filters1.10 libpcl-io1.10 libpcl-kdtree1.10 libpcl-keypoints1.10
  libpcl-ml1.10 libpcl-octree1.10 libpcl-outofcore1.10 libpcl-people1.10 libpcl-recognition1.10 libpcl-registration1.10 libpcl-sample-consensus1.10 libpcl-search1.10 libpcl-segmentation1.10 libpcl-stereo1.10 libpcl-surface1.10
  libpcl-tracking1.10 libpcl-visualization1.10 libpcre2-16-0 libpoppler-dev libpoppler-private-dev libpq-dev libproj-dev libpthread-stubs0-dev libqhull-dev libqhull-r7 libqt5concurrent5 libqt5core5a libqt5dbus5 libqt5designer5
  libqt5designercomponents5 libqt5gui5 libqt5help5 libqt5network5 libqt5opengl5 libqt5opengl5-dev libqt5positioning5 libqt5printsupport5 libqt5qml5 libqt5quick5 libqt5quickparticles5 libqt5quickshapes5 libqt5quicktest5
  libqt5quickwidgets5 libqt5sensors5 libqt5sql5 libqt5sql5-sqlite libqt5test5 libqt5webchannel5 libqt5webkit5 libqt5webkit5-dev libqt5widgets5 libqt5x11extras5 libqt5xml5 libsm-dev libspatialite-dev libsuperlu-dev libtheora-dev
  libtk8.6 liburiparser-dev libvtk7-dev libvtk7-java libvtk7-jni libvtk7-qt-dev libvtk7.1p libvtk7.1p-qt libvulkan-dev libwacom-common libwacom2 libwayland-server0 libwebp-dev libwoff1 libx11-dev libxau-dev libxcb-icccm4
  libxcb-image0 libxcb-keysyms1 libxcb-render-util0 libxcb-shape0 libxcb-util1 libxcb-xinerama0 libxcb-xinput0 libxcb-xkb1 libxcb1-dev libxdmcp-dev libxerces-c-dev libxext-dev libxft-dev libxft2 libxkbcommon-x11-0 libxml2-dev
  libxrender-dev libxss-dev libxt-dev libzstd-dev python3-autobahn python3-cbor python3-lz4 python3-mpi4py python3-pyqrcode python3-snappy python3-trie python3-txaio python3-u-msgpack python3-ubjson python3-vtk7 python3-wsaccel
  qdoc-qt5 qhelpgenerator-qt5 qt5-assistant qt5-qmake qt5-qmake-bin qt5-qmltooling-plugins qtattributionsscanner-qt5 qtbase5-dev qtbase5-dev-tools qtchooser qtdeclarative5-dev qtdeclarative5-dev-tools qttools5-dev
  qttools5-dev-tools qttools5-private-dev ros-noetic-pcl-conversions ros-noetic-pcl-msgs ros-noetic-tf2-eigen tcl tcl-dev tcl-vtk7 tcl8.6-dev tk tk-dev tk8.6 tk8.6-dev unixodbc-dev vtk7 x11proto-core-dev x11proto-dev
  x11proto-scrnsaver-dev x11proto-xext-dev xorg-sgml-doctools xtrans-dev
Suggested packages:
  cython-doc libitpp-dev liblapack-doc libcurl4-doc libgnutls28-dev libidn11-dev libkrb5-dev librtmp-dev libssh2-1-dev freetype2-doc libgdal-doc libgeotiff-epsg glew-utils libhdf5-doc libice-doc netcdf-bin netcdf-doc openni-doc
  openni2-doc libpcl-doc postgresql-doc-12 qt5-image-formats-plugins qtwayland5 libsm-doc libsuperlu-doc vtk7-doc vtk7-examples java-virtual-machine libx11-doc libxcb-doc libxerces-c-doc libxext-doc libxt-doc python-trie-doc
  python-txaio-doc mayavi2 qt5-doc firebird-dev tcl-doc tcl8.6-doc tk-doc tk8.6-doc
Recommended packages:
  libcfitsio-doc libopenni-sensor-pointclouds0 | libopenni-sensor-primesense0 qttranslations5-l10n libqt5svg5 qt5-gtk-platformtheme geoclue-2.0 libwacom-bin python3-png xterm | x-terminal-emulator
The following NEW packages will be installed:
  cython3 hdf5-helpers libaec-dev libarmadillo-dev libarpack2-dev libblas-dev libcfitsio-dev libcharls-dev libclang1-10 libcurl4-gnutls-dev libdap-dev libdapserver7v5 libdouble-conversion3 libegl-dev libegl-mesa0 libegl1
  libepsilon-dev libflann-dev libflann1.9 libfontconfig1-dev libfreetype-dev libfreetype6-dev libfreexl-dev libfyba-dev libgbm1 libgdal-dev libgeos-dev libgeotiff-dev libgif-dev libgl-dev libgl1-mesa-dev libgl2ps-dev
  libgles-dev libgles1 libgles2 libglew-dev libglew2.1 libglu1-mesa libglu1-mesa-dev libglvnd-dev libglx-dev libhdf4-alt-dev libhdf5-cpp-103 libhdf5-dev libhdf5-mpi-dev libhdf5-openmpi-dev libhyphen0 libice-dev libinput-bin
[... cut ...]
  qttools5-dev-tools qttools5-private-dev ros-noetic-pcl-conversions ros-noetic-pcl-msgs ros-noetic-pcl-ros ros-noetic-tf2-eigen tcl tcl-dev tcl-vtk7 tcl8.6-dev tk tk-dev tk8.6 tk8.6-dev unixodbc-dev vtk7 x11proto-core-dev
  x11proto-dev x11proto-scrnsaver-dev x11proto-xext-dev xorg-sgml-doctools xtrans-dev
```

At the very least using the `--no-install-recommends` flag saves ~200MB of dependencies from being brought in.

[apt-cache.txt](https://github.com/ros-perception/perception_pcl/files/7974158/apt-cache.txt)
[apt-rdepends.txt](https://github.com/ros-perception/perception_pcl/files/7974159/apt-rdepends.txt)

JanezCim, 2022-04-26:
+1, why would you need full `gdm3` to be installed with this package?

parameciumvance, 2022-04-29:
I have researched this problem.
`apt-rdepends -f Depends,Recommends -s Depends,Recommends ros-noetic-pcl-ros`
Using this command, we can locate what brings  gnome-shell (and hence gdm3).

The key part is ...
"libnotify4" which has gnome-shell as Recommended dependency

[https://packages.ubuntu.com/focal/libnotify4](url)
From Ubuntu's package description, we can find that, gnome-shell has been added to  libnotify4's Recommended dependency in focal (Ubuntu20.04), resulting that, when you install libnotify4 in some way, you also install gnome-shell

Also notice that in jammy (Ubuntu22.04), gnome-shell in dependency has dropped from Recommended to Suggested, I think the package manager may have been aware of the problem and fixed it then.

My suggestion is that, when using Ubuntu server (20.04<=version<22.04),
always manually install the problematic libnotify4 in early setting
`sudo apt install libnotify4 --no-install-recommends`
This prevents unintentionally install gnome-shell as dependency afterward.

ps. not sure if there is other libs besides libnotify4 has the same property. If this happens again, just use the same method to locate the problematic lib.

civerachb-cpr, 2022-05-10:
Work-Around Solution
==================

Install `notify-osd` with apt before installing `ros-noetic-pcl-ros`:
```
sudo apt-get install notify-osd
sudo apt-get install ros-noetic-pcl-ros
```

Background
===========

Thanks to @mhosmar-cpr and David G from the VTK team for digging into this more.  After determining that the issue didn't appear to be with the PCL packages themselves, our next suspect was VTK, which is included as a dependency.  We opened a ticket on their GitLab page, and one of the developers there was able to dig into the issue more and identified the culprit as an altered dependency of `ffmpeg`:

The issue appears to have been caused by the `notification-daemon` virtual package being a necessary dependency of ffmpeg:
> I dug just a bit deeper, and found that the focal vs. xenial change occurred within ffmpeg.
> The libavutil56 package requires libopencl1, whereas the older libavutil55 package did not.
> -[source](https://gitlab.kitware.com/vtk/vtk/-/issues/18489#note_1156770)

If the dependency on `libopencl1` is not satisfied prior to installing `ros-noetic-pcl-ros`, apt pulls in the recommended defaults to provide that dependency.  This causes the Ubuntu desktop, GDM, and other unwanted GUI packages to be pulled in as they're the defaults.

Installing `notify-osd` first appears to resolve the dependency and prevents the additional graphical packages from being pulled-in.

This works even if apt is configured to install recommended packages.  (Note that on a command-line only system `pcl_ros` still pulls in a LOT of other dependencies, but you should at least be able to avoid having GDM suddenly present on your previously headless system.)