ros-perception/perception_pcl #364: Using pcl_ros::transformPointCloud<pointT> with own pointT: Linker Error

Scyten, 2022-05-05:
Hi Community,
I have an issue using this function
```
template<typename PointT >
bool pcl_ros::transformPointCloud (const std::string &target_frame, const pcl::PointCloud< PointT > &cloud_in, pcl::PointCloud< PointT > &cloud_out, const tf::TransformListener &tf_listener)
```
If I use it with a "standard" pointT-type it works fine.
If I use my own datatype it compiles, but I get undefined reference to the function above (pcl_ros::transformPointCloud) while linking.

PCL requires 
`#define PCL_NO_PRECOMPILE`
to work with new point types (https://pointclouds.org/documentation/tutorials/adding_custom_ptype.html)

My pointT is
```
struct mmWaveCloudType
{
    PCL_ADD_POINT4D;
    union
    {
        struct
        {
            float intensity;
            float velocity;
        };
        float data_c[4];
    };
    EIGEN_MAKE_ALIGNED_OPERATOR_NEW
} EIGN_ALIGN16;

POINT_CLOUD_REGISTER_POINT_STRUCT (mmWaveCloudType,
                                    (float, x, x)
                                    (float, y, y)
                                    (float, z, z)
                                    (float, intensity, intensity)
                                    (float, velocity, velocity))
```

Is this define used by ros_pcl? 
Has someone tested new pointTypes with ros_pcl?

mvieth, 2022-05-06:
Have you tried including `pcl_ros/impl/transforms.hpp`? That's where the implementation is

mvieth, 2022-05-07:
@Scyten `pcl_ros/transforms.h` only contains the declaration of the templated function. `pcl_ros/impl/transforms.hpp` contains the implementation. If the implementation is available to your compiler, it can instantiate it with any point type.