ros-perception/perception_pcl #366: Converting Point cloud to image fails without RGB channel

RMichaelSwan, 2022-05-11:
The `convert_pointcloud_to_image.cpp` tool does not work for many common point clouds. It depends on the point cloud including an RGB field, which would only make sense for RGB-D sensors. 

It looks like this may be a fundamental issue with the pcl library not supporting this type of conversion. Source of the bug: https://github.com/ros-perception/perception_pcl/blob/bd7a060e98d2bb6720b3bc30794fd25c5ef43241/pcl_conversions/include/pcl_conversions/pcl_conversions.h#L521
https://github.com/PointCloudLibrary/pcl/blob/83eecf88f56b9e6812d8875bfb30e1140e01dd5a/common/include/pcl/conversions.h#L316

I have found some converters people rolled themselves, though they are a bit old, so will take some work to get working on ROS Noetic:
https://github.com/v-lopez/pointcloud_to_rangeimage/tree/srcsim-fixes
https://github.com/EPVelasco/pc2image

Petros626, 2022-05-13:
Hey,

its the same problem for me. I tried exactly the same script, but the PointCloud type isn't suitable. Did you made it running with one of these nodes?

RMichaelSwan, 2022-05-18:
> Hey,
> 
> its the same problem for me. I tried exactly the same script, but the PointCloud type isn't suitable. Did you made it running with one of these nodes? I would suggest you another solution, which we maybe can figure out together? Look here: https://stackoverflow.com/questions/52481774/how-to-save-intensity-value-in-sensor-msgs-image-from-pointcloud https://answers.ros.org/question/9515/how-to-convert-between-different-point-cloud-types-using-pcl/

I ended up using this repo inside of a ROS Kinetic docker container and recording a rosbag of the output. Unfortunately it does not work with ROS Noetic as the pcl library changed some things since this code was written. Could probably do a rewrite, but I don't feel the urge to get into it right now. Relevant repo: https://github.com/EPVelasco/pc2image