ros-perception/perception_pcl #367: error compiling pcl::toROSMsg and pcl::fromROSMsg with Humble on Ubuntu Jammy

hortovanyi, 2022-05-18:
Previously this code ([public lidar driver code here](https://github.com/aussierobots/rs_ros2_lidar)) worked fine on ROS2 Galactic on Ubuntu 20.04. Now with ROS2 Humble on Ubuntu 22.04 it no longer compiles giving the following errors

Unable to work out how to fix this templating issue. Error log follows

/home/nick/sdcar-poc5-ws/src/rs_ros2_lidar/src/lidar_pub.cpp: In member function ‘void rs_lidar::LIDARPub::publish_point_cloud_from_msg(const robosense::lidar::PointCloudMsg<pcl::PointXYZI>&)’:
/home/nick/sdcar-poc5-ws/src/rs_ros2_lidar/src/lidar_pub.cpp:161:18: error: no matching function for call to ‘toROSMsg(pcl::PointCloud<pcl::PointXYZI>&, std::__shared_ptr_access<sensor_msgs::msg::PointCloud2_<std::allocator<void> >, __gnu_cxx::_S_atomic, false, false>::element_type&)’
  161 |     pcl::toROSMsg(point_cloud, *pc2_msg);
      |     ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/nick/sdcar-poc5-ws/src/rs_ros2_lidar/src/lidar_pub.cpp:4:
/usr/include/pcl_conversions/pcl_conversions.h:521:3: note: candidate: ‘template<class T> void pcl::toROSMsg(const pcl::PointCloud<PointT>&, sensor_msgs::Image&)’
  521 |   toROSMsg (const pcl::PointCloud<T> &cloud, sensor_msgs::Image& msg)
      |   ^~~~~~~~
/usr/include/pcl_conversions/pcl_conversions.h:521:3: note:   template argument deduction/substitution failed:
/home/nick/sdcar-poc5-ws/src/rs_ros2_lidar/src/lidar_pub.cpp:161:32: note:   cannot convert ‘((std::__shared_ptr_access<sensor_msgs::msg::PointCloud2_<std::allocator<void> >, __gnu_cxx::_S_atomic, false, false>*)(& pc2_msg))->std::__shared_ptr_access<sensor_msgs::msg::PointCloud2_<std::allocator<void> >, __gnu_cxx::_S_atomic, false, false>::operator*()’ (type ‘std::__shared_ptr_access<sensor_msgs::msg::PointCloud2_<std::allocator<void> >, __gnu_cxx::_S_atomic, false, false>::element_type’ {aka ‘sensor_msgs::msg::PointCloud2_<std::allocator<void> >’}) to type ‘sensor_msgs::Image&’ {aka ‘sensor_msgs::Image_<std::allocator<void> >&’}
  161 |     pcl::toROSMsg(point_cloud, *pc2_msg);
      |                                ^~~~~~~~
In file included from /home/nick/sdcar-poc5-ws/src/rs_ros2_lidar/src/lidar_pub.cpp:4:
/usr/include/pcl_conversions/pcl_conversions.h:553:8: note: candidate: ‘template<class T> void pcl::toROSMsg(const pcl::PointCloud<PointT>&, sensor_msgs::PointCloud2&)’
  553 |   void toROSMsg(const pcl::PointCloud<T> &pcl_cloud, sensor_msgs::PointCloud2 &cloud)
      |        ^~~~~~~~
/usr/include/pcl_conversions/pcl_conversions.h:553:8: note:   template argument deduction/substitution failed:
/home/nick/sdcar-poc5-ws/src/rs_ros2_lidar/src/lidar_pub.cpp:161:32: note:   cannot convert ‘((std::__shared_ptr_access<sensor_msgs::msg::PointCloud2_<std::allocator<void> >, __gnu_cxx::_S_atomic, false, false>*)(& pc2_msg))->std::__shared_ptr_access<sensor_msgs::msg::PointCloud2_<std::allocator<void> >, __gnu_cxx::_S_atomic, false, false>::operator*()’ (type ‘std::__shared_ptr_access<sensor_msgs::msg::PointCloud2_<std::allocator<void> >, __gnu_cxx::_S_atomic, false, false>::element_type’ {aka ‘sensor_msgs::msg::PointCloud2_<std::allocator<void> >’}) to type ‘sensor_msgs::PointCloud2&’ {aka ‘sensor_msgs::PointCloud2_<std::allocator<void> >&’}
  161 |     pcl::toROSMsg(point_cloud, *pc2_msg);
      |                                ^~~~~~~~
In file included from /home/nick/sdcar-poc5-ws/src/rs_ros2_lidar/src/lidar_pub.cpp:4:
/usr/include/pcl_conversions/pcl_conversions.h:501:8: note: candidate: ‘void pcl::toROSMsg(const PointCloud2&, sensor_msgs::Image&)’
  501 |   void toROSMsg(const sensor_msgs::PointCloud2 &cloud, sensor_msgs::Image &image)
      |        ^~~~~~~~
/usr/include/pcl_conversions/pcl_conversions.h:501:49: note:   no known conversion for argument 1 from ‘pcl::PointCloud<pcl::PointXYZI>’ to ‘const PointCloud2&’ {aka ‘const sensor_msgs::PointCloud2_<std::allocator<void> >&’}
  501 |   void toROSMsg(const sensor_msgs::PointCloud2 &cloud, sensor_msgs::Image &image)
      |                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/home/nick/sdcar-poc5-ws/src/rs_ros2_lidar/src/lidar_viz.cpp: In lambda function:
/home/nick/sdcar-poc5-ws/src/rs_ros2_lidar/src/lidar_viz.cpp:44:28: error: no matching function for call to ‘fromROSMsg(std::__shared_ptr_access<const sensor_msgs::msg::PointCloud2_<std::allocator<void> >, __gnu_cxx::_S_atomic, false, false>::element_type&, std::__shared_ptr_access<pcl::PointCloud<pcl::PointXYZI>, __gnu_cxx::_S_atomic, false, false>::element_type&)’
   44 |             pcl::fromROSMsg(*pc2_msg, *inputCloudI);
      |             ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/nick/sdcar-poc5-ws/src/rs_ros2_lidar/src/lidar_viz.cpp:4:
/usr/include/pcl_conversions/pcl_conversions.h:561:8: note: candidate: ‘template<class T> void pcl::fromROSMsg(const PointCloud2&, pcl::PointCloud<PointT>&)’
  561 |   void fromROSMsg(const sensor_msgs::PointCloud2 &cloud, pcl::PointCloud<T> &pcl_cloud)
      |        ^~~~~~~~~~
/usr/include/pcl_conversions/pcl_conversions.h:561:8: note:   template argument deduction/substitution failed:
/home/nick/sdcar-poc5-ws/src/rs_ros2_lidar/src/lidar_viz.cpp:44:29: note:   cannot convert ‘((std::__shared_ptr_access<const sensor_msgs::msg::PointCloud2_<std::allocator<void> >, __gnu_cxx::_S_atomic, false, false>*)(& pc2_msg))->std::__shared_ptr_access<const sensor_msgs::msg::PointCloud2_<std::allocator<void> >, __gnu_cxx::_S_atomic, false, false>::operator*()’ (type ‘std::__shared_ptr_access<const sensor_msgs::msg::PointCloud2_<std::allocator<void> >, __gnu_cxx::_S_atomic, false, false>::element_type’ {aka ‘const sensor_msgs::msg::PointCloud2_<std::allocator<void> >’}) to type ‘const PointCloud2&’ {aka ‘const sensor_msgs::PointCloud2_<std::allocator<void> >&’}
   44 |             pcl::fromROSMsg(*pc2_msg, *inputCloudI);
      |                             ^~~~~~~~

mvieth, 2022-05-19:
How did you install perception_pcl/pcl_conversions? If you compiled it from source, which branch did you use?

hortovanyi, 2022-05-19:
Installed it via ros-humbe package … its available

mvieth, 2022-05-19:
Can you post the exact command you used to install it?
The error messages mention `/usr/include/pcl_conversions/pcl_conversions.h`. I don't have much experience with ROS2, but shouldn't the ROS stuff be in `/opt/ros/humble/...` ? Can you check whether `pcl_conversions.h` exists more than once on your computer?

hortovanyi, 2022-05-19:
It’s all installed. Can see that it’s a templating issue I think from the error messages. Something to do with alloc?

I haven’t got the machine turned on. I did it through sudo aptitude 

I would of sourced /opt/ros/humble/setup.zsh

conda deactivate
rm -rf install build log
colcon build —symlink-install

mvieth, 2022-05-19:
Of the 4 possible candidates listed in the error message, the second candidate should fit (`template void pcl::toROSMsg(const pcl::PointCloud&, sensor_msgs::PointCloud2&)`). A type mismatch between `sensor_msgs::msg::PointCloud2<std::allocator >` and `sensor_msgs::PointCloud2<std::allocator >&` is reported. Considering the missing `msg` between `sensor_msgs` and `PointCloud2` in the latter type, as well as the line number of the possible candidates, it seems like you have `pcl_conversions.h` from the melodic-devel branch installed under `/usr/include/pcl_conversions` (the melodic branch is also used in the noetic distribution). Even if you have the correct package from humble installed, the wrong header seems to take precedence.

SteveMacenski, 2022-05-19:
I've also experienced some ROS 1/ROS 2 side-by-side install issues with specifically headers, I'd try isolating that first too.

hortovanyi, 2022-05-19:
Thanks @SteveMacenski @mvieth  ran 'sudo apt remove libpcl-ros-dev pcl-ros-tools'  and then 'sudo apt autoremove' and now am able to build.

Also had to add pcl_conversions to ament_target_dependencies 
`ament_target_dependencies(rs_lidar_components
  rclcpp
  rclcpp_components
  sensor_msgs
  pcl_conversions
)`

closed 2022-05-19 by hortovanyi