ros-perception/perception_pcl #387: Latest PCL Version (1.12) Not Compatible with perception_pcl

AmirpooyaSh, 2022-12-08:
Dear All.

I was recently trying to use filtering parts of pcl library (Frustum_Culling.h) on Ros-Melodic.
However, the Melodic version will accept pcl 1.8 which doesn't have the required fuctions I'm working on.
Therefore, I tried uninstalling libpcl-dev and building pcl 1.12 from their github repository. Suprisingly I recieved a lot of errors related to conflicts between perception_pcl and pcl.

I was trying to replace the required headers from pcl 1.12 to 1.8 (Frustum_Culling.h, Memory.h, Point_Traits.h, Point_Struct_Traits.h), but it seems there is a lot of old functions written in perception_pcl library that can't get connected to newer versions of PCL.

As I've read PCL github issues, they're saying that PCL is a seperate library from ROS and they don't know why the perception_pcl is not working with the newer versions of PCL. I think that these ros library haven't been updated for a long time or maybe got updated in ros-neotic or even ros2 ?

For conclusion, I was wondering if you could give me any kind of alternatives through my problem or hard coding the part I'm going to need is my only option.

mvieth, 2022-12-08:
Hi, PCL maintainer here

> As I've read PCL github issues, they're saying that PCL is a seperate library from ROS

That is correct, PCL is independent from and can be used without ROS. `perception_pcl` (and `pcl_conversions`, `pcl_ros`) are the bridge between PCL and ROS

> and they don't know why the perception_pcl is not working with the newer versions of PCL

No idea where you got this from, can you link the GitHub issue where you read this?

`perception_pcl` is regularly kept up-to-date to work with the latest PCL releases. If you use ROS Melodic, you just need compile from source using the `melodic-devel` branch (for ROS Noetic you would BTW also use the `melodic-devel` branch). If this for some reason does not work, please post the error messages.

AmirpooyaSh, 2022-12-08:
I got a bit confused with the concept melodic-devel you are talking about. It seems to be dummy, but could you tell me how can I compile the perception_pcl using devel branch of melodic.

I've already gone through the industrial core tutorial. I believe that the industrial core is something like anaconda to compile packages with different versions togather, but I have no idea how to work with it. I would appreciate it if you can send any tutorial for using it or give me some hints.

mvieth, 2022-12-09:
[These tutorials](https://wiki.ros.org/ROS/Tutorials) could be a starting point for you, especially "Building a ROS Package"

closed 2022-12-09 by SteveMacenski

AmirpooyaSh, 2022-12-13:
I can't find something related to melodic-devel branch indeed.
What I've done in ROS so far was to use the formal way of installing ros-melodic as it's been described in the [LINK](http://wiki.ros.org/melodic/Installation/Ubuntu) and using the installed libraries to create my custom packages and make it work.

Now I got into the trouble of an old version library. When I install ros-melodic on Ubuntu 18.04, it will install PCL-1.8 as its dependency.
I'm trying to use PCL/FILTERS/FRUSTUM_CULLING.H part of PCL library.
There are some changes in PCL-1.12 comparing to PCL-1.8 to this header file (newly functions and operators have been written).
What I've found out from your response was to create a new workspace, build melodic-devel as a package in it and source it to compile my custom package, but how ?

https://github.com/ros/ros/tree/melodic-devel#readme
This was all I've found for melodic-devel without any guidance on working with it neither on the ROS basic tutorials.

I can't understand the way of working with melodic-devel. Thats my problem and I couldn't find any direct tutorial to it, so could you give me some more details about that.

 @mvieth

mvieth, 2022-12-13:
@AmirpooyaSh What I meant is that you should use the `melodic-devel` branch of this repository (`perception_pcl`) when you compile it from source. So `git checkout melodic-devel` after you clone this repository. The default branch of this repository is for ROS2, which won't work if you use ROS1 (e.g. ROS Melodic)

AmirpooyaSh, 2022-12-13:
@mvieth Ow......! Thanks for the clarification.
I did what you've meant, but when I'm trying to catkin_make the package it gives me errors related to a function called BOOST_FOREACH which has not been declared in point_cloud.h file.

In file included from /home/kyungki/ros_pcl/src/pcl_ros/src/pcl_ros/surface/convex_hull.cpp:39:0:
/home/kyungki/ros_pcl/src/pcl_ros/include/pcl_ros/surface/convex_hull.h:44:10: fatal error: pcl/surface/convex_hull.h: No such file or directory
 #include <pcl/surface/convex_hull.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
pcl_ros/CMakeFiles/pcl_ros_surface.dir/build.make:89: recipe for target 'pcl_ros/CMakeFiles/pcl_ros_surface.dir/src/pcl_ros/surface/convex_hull.cpp.o' failed
make[2]: *** [pcl_ros/CMakeFiles/pcl_ros_surface.dir/src/pcl_ros/surface/convex_hull.cpp.o] Error 1
CMakeFiles/Makefile2:3071: recipe for target 'pcl_ros/CMakeFiles/pcl_ros_surface.dir/all' failed
make[1]: *** [pcl_ros/CMakeFiles/pcl_ros_surface.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 35%] Building CXX object pcl_ros/CMakeFiles/pcl_ros_features.dir/src/pcl_ros/features/fpfh_omp.cpp.o
[ 37%] Linking CXX executable /home/kyungki/ros_pcl/devel/lib/pcl_ros/convert_pcd_to_image
In file included from /home/kyungki/ros_pcl/src/pcl_ros/include/pcl_ros/point_cloud.h:6:0,
                 from /home/kyungki/ros_pcl/src/pcl_ros/include/pcl_ros/pcl_nodelet.h:54,
                 from /home/kyungki/ros_pcl/src/pcl_ros/include/pcl_ros/features/feature.h:44,
                 from /home/kyungki/ros_pcl/src/pcl_ros/include/pcl_ros/features/fpfh.h:42,
                 from /home/kyungki/ros_pcl/src/pcl_ros/src/pcl_ros/features/fpfh.cpp:39:
/usr/local/include/pcl-1.13/pcl/point_traits.h:41:64: note: #pragma message: This header is deprecated. Use <pcl/type_traits.h> instead. (It will be removed in PCL 1.15)
 PCL_DEPRECATED_HEADER(1, 15, "Use <pcl/type_traits.h> instead.")
                                                                ^
/usr/local/include/pcl-1.13/pcl/point_traits.h:41:64: note: #pragma message: This header is deprecated. Use <pcl/type_traits.h> instead. (It will be removed in PCL 1.15)
/usr/local/include/pcl-1.13/pcl/point_traits.h:41:64: note: #pragma message: This header is deprecated. Use <pcl/type_traits.h> instead. (It will be removed in PCL 1.15)
/usr/local/include/pcl-1.13/pcl/point_traits.h:41:64: note: #pragma message: This header is deprecated. Use <pcl/type_traits.h> instead. (It will be removed in PCL 1.15)
In file included from /home/kyungki/ros_pcl/src/pcl_ros/include/pcl_ros/pcl_nodelet.h:54:0,
                 from /home/kyungki/ros_pcl/src/pcl_ros/include/pcl_ros/features/feature.h:44,
                 from /home/kyungki/ros_pcl/src/pcl_ros/include/pcl_ros/features/fpfh.h:42,
                 from /home/kyungki/ros_pcl/src/pcl_ros/src/pcl_ros/features/fpfh.cpp:39:
/home/kyungki/ros_pcl/src/pcl_ros/include/pcl_ros/point_cloud.h: In static member function ‘static void ros::serialization::Serializer<pcl::PointCloud<PointT> >::read(Stream&, pcl::PointCloud<PointT>&)’:
/home/kyungki/ros_pcl/src/pcl_ros/include/pcl_ros/point_cloud.h:231:29: error: expected primary-expression before ‘const’
               BOOST_FOREACH(const pcl::detail::FieldMapping& fm, mapping) {
                             ^~~~~
/home/kyungki/ros_pcl/src/pcl_ros/include/pcl_ros/point_cloud.h:231:15: error: there are no arguments to ‘BOOST_FOREACH’ that depend on a template parameter, so a declaration of ‘BOOST_FOREACH’ must be available [-fpermissive]
               BOOST_FOREACH(const pcl::detail::FieldMapping& fm, mapping) {
               ^~~~~~~~~~~~~
/home/kyungki/ros_pcl/src/pcl_ros/include/pcl_ros/point_cloud.h:231:15: note: (if you use ‘-fpermissive’, G++ will accept your code, but allowing the use of an undeclared name is deprecated)
/home/kyungki/ros_pcl/src/pcl_ros/include/pcl_ros/point_cloud.h:231:75: error: expected ‘;’ before ‘{’ token
               BOOST_FOREACH(const pcl::detail::FieldMapping& fm, mapping) {
                                                                           ^
In file included from /home/kyungki/ros_pcl/src/pcl_ros/include/pcl_ros/point_cloud.h:6:0,
                 from /home/kyungki/ros_pcl/src/pcl_ros/include/pcl_ros/pcl_nodelet.h:54,
                 from /home/kyungki/ros_pcl/src/pcl_ros/include/pcl_ros/features/feature.h:44,
                 from /home/kyungki/ros_pcl/src/pcl_ros/src/pcl_ros/features/feature.cpp:50:
/usr/local/include/pcl-1.13/pcl/point_traits.h:41:64: note: #pragma message: This header is deprecated. Use <pcl/type_traits.h> instead. (It will be removed in PCL 1.15)
 PCL_DEPRECATED_HEADER(1, 15, "Use <pcl/type_traits.h> instead.")
                                                                ^
/usr/local/include/pcl-1.13/pcl/point_traits.h:41:64: note: #pragma message: This header is deprecated. Use <pcl/type_traits.h> instead. (It will be removed in PCL 1.15)
/usr/local/include/pcl-1.13/pcl/point_traits.h:41:64: note: #pragma message: This header is deprecated. Use <pcl/type_traits.h> instead. (It will be removed in PCL 1.15)
/usr/local/include/pcl-1.13/pcl/point_traits.h:41:64: note: #pragma message: This header is deprecated. Use <pcl/type_traits.h> instead. (It will be removed in PCL 1.15)
[ 39%] Linking CXX executable /home/kyungki/ros_pcl/devel/lib/pcl_ros/pcd_to_pointcloud
In file included from /home/kyungki/ros_pcl/src/pcl_ros/include/pcl_ros/pcl_nodelet.h:54:0,
[... cut ...]
pcl_ros/CMakeFiles/pcl_ros_features.dir/build.make:131: recipe for target 'pcl_ros/CMakeFiles/pcl_ros_features.dir/src/pcl_ros/features/shot.cpp.o' failed
make[2]: *** [pcl_ros/CMakeFiles/pcl_ros_features.dir/src/pcl_ros/features/shot.cpp.o] Error 1
CMakeFiles/Makefile2:2963: recipe for target 'pcl_ros/CMakeFiles/pcl_ros_features.dir/all' failed
make[1]: *** [pcl_ros/CMakeFiles/pcl_ros_features.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

mvieth, 2022-12-13:
@AmirpooyaSh You have to use the **current** melodic-devel branch. The code you are using seems to be a few months old. So pull the newest changes from GitHub.

AmirpooyaSh, 2022-12-13:
@mvieth Sorry for bothering you that much. I was trying to find a way for 7 days and now I think that I got close to the solution.
I cloned the newest melodic-devel branch and just received this error.

In file included from /home/kyungki/ros_pcl/src/perception_pcl/pcl_ros/src/pcl_ros/surface/convex_hull.cpp:39:0:
/home/kyungki/ros_pcl/src/perception_pcl/pcl_ros/include/pcl_ros/surface/convex_hull.h:44:10: fatal error: pcl/surface/convex_hull.h: No such file or directory
 #include <pcl/surface/convex_hull.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
perception_pcl/pcl_ros/CMakeFiles/pcl_ros_surface.dir/build.make:89: recipe for target 'perception_pcl/pcl_ros/CMakeFiles/pcl_ros_surface.dir/src/pcl_ros/surface/convex_hull.cpp.o' failed
make[2]: *** [perception_pcl/pcl_ros/CMakeFiles/pcl_ros_surface.dir/src/pcl_ros/surface/convex_hull.cpp.o] Error 1
CMakeFiles/Makefile2:3071: recipe for target 'perception_pcl/pcl_ros/CMakeFiles/pcl_ros_surface.dir/all' failed
make[1]: *** [perception_pcl/pcl_ros/CMakeFiles/pcl_ros_surface.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

Isn't it something because of their recent PCL-1.13 update ?
At this point I do have PCL-1.13 manually built on my computer, so I don't know why it can't find that header file in PCL library !

AmirpooyaSh, 2022-12-13:
@mvieth 
To be honest, there were a lot of dependencies needed to be upgrade along side PCL in order to work with the latest version of PERCEPTION_PCL for ros-melodic.
Thank you for your help. I finally could start using PCL-1.13 with ROS-Melodic ;).