ros-perception/perception_pcl #396: ROS2 looking for different pcl version when compiling

YouBotica, 2023-02-02:
Hi all.
I am working with the pcl library in ROS2 Galactic, I am trying to upgrade my pcl library from 1.10 to 1.13 to have the latest version and run my lidar nodes with gpu support (pcl::gpu...). I followed the official pcl tutorials for installing pcl::cuda (https://pcl.readthedocs.io/projects/tutorials/en/pcl-1.11.0/gpu_install.html) and built pcl1.13 from source. However, when I compile my packages that use pcl using colcon build, the compiler attempts to look for the header files in the old version's folder (which no longer exists) rather than the new one. How can I fix this? Is pcl 1.13 compatible with ros2 galactic and ros-galactic-pcl* pkgs? Where can I find more information on upgrading pcl along with ros2 and compatibility?

Thanks

mvieth, 2023-02-02:
@YouBotica Have you removed pcl-conversions, pcl-ros, and pcl-perception (the packages installed via apt), and built them from source instead?

YouBotica, 2023-02-02:
I am building those packages from source, but haven't removed them from apt, I thought that when you build them from source these were overwritten. When I try to do: sudo apt remove ros-galactic-pcl-conversions it asks me to also uninstall ros-galactic-desktop and other important ros galactic filed which I definitely don't want to get rid of. It seems like when I try to uninstall ros-galactic-pcl* pkgs it proceeds to uninstall core libraries of ros2 galactic, I don't know why.

mvieth, 2023-02-02:
I believe ros-galactic-desktop is just an alias for a collection of frequently used packages. If you remove ros-galactic-desktop, it shouldn't remove any other packages immediately, only if you also do auto-remove. I am not aware of any "core libraries" that depend on ros-galactic-pcl-conversions.
This is what I believe is the case, but please check the apt output to know for sure.

YouBotica, 2023-02-04:
Gotcha! Thank you! 

The issue was that some packages were not building due to errors, I changed my compiler to gcc 9 as default and the compilation went with no errors. It is weird because in the documentation (https://pcl.readthedocs.io/projects/tutorials/en/pcl-1.13.0/gpu_install.html) it says that gcc 7 should be used, but that compiler was giving a lot of errors. I also added these lines to the CMAKE file of my ROS2 node using PCL::GPU:

find_package(PCL 1.13 REQUIRED)

....

include_directories(include ${PCL_INCLUDE_DIRS})
link_directories(${PCL_LIBRARY_DIRS})
add_definitions(${PCL_DEFINITIONS})

# Link the latest pcl 1.13 library installed from source:
set(LD_LIBRARY_PATH ${LD_LIBRARY_PATH} /usr/local/lib)
link_directories(/usr/local/lib)

However, now I am having issues with apparently my Cuda version. I am getting the error:

ros2: Error: invalid device function	/home/ai/Downloads/pcl/gpu/octree/src/cuda/octree_host.cu:65:get_gpu_arch_compiled_for

I am using Cuda 11.7, Cuda toolkit 10, and PCL 1.13 in ROS2 Galactic. 
I have not been able to find information about if Cuda 11.7 is compatible with PCL 1.13 (Maybe I am not looking correctly for the info). Any ideas on how to solve this?

foxydevelby, 2023-05-26:
Hello @YouBotica, have you successfully use pcl_ros in galactic?, I also cannot use it.