ros-perception/perception_pcl #403: 1.7.4 does not include the fix the missing BOOST FOREACH macro

mob41, 2023-02-15:
Hi, I was compiling `ros-noetic-desktop-full` in my arch linux installation. And found that the latest 1.7.4 release does not include the fix https://github.com/ros-perception/perception_pcl/commit/85334c39256530cf27abf6accbc7b55e18a2da02  for the missing BOOST FOREACH macro, causing me unable to compile this package.

```
/home/anth/.cache/paru/clone/ros-noetic-pcl-ros/src/perception_pcl-1.7.4/pcl_ros/include/pcl_ros/point_cloud.h:231:15: error: there are no arguments to ‘BOOST_FOREACH’ that depend on a template parameter, so a declaration of ‘BOOST_FOREACH’ must be available [-fpermissive]
  231 |               BOOST_FOREACH(const pcl::detail::FieldMapping& fm, mapping) {
      |               ^~~~~~~~~~~~~
```

https://github.com/ros-perception/perception_pcl/blob/1.7.4/pcl_ros/include/pcl_ros/point_cloud.h

mvieth, 2023-05-11:
@SteveMacenski Are there still plans for a release?

SteveMacenski, 2023-05-18:
It looks like @paulbovbel made a release for 1.7.5 about 4 months ago. Should/can that be released into bloom Paul?