ros-perception/perception_pcl #409: PCL Octree error in catkin_make

AnOrdinaryUsser, 2023-03-22:
Good,

I'm trying to make my robot, which uses the D415 camera from intel, to be able to detect obstacles thanks to the point cloud and then I can use that information to make it able to avoid them. 
I am using the following:
- intel realsense 2 package for the camera
- ROS Melodic
- Ubuntu 18.04 
- Jetson Nano (ARM)

And the following pcl packages:
```bash
libdapclient6v5/bionic,now 3.19.1-2build1 arm64 [instalado, automático]
libpcl-apps1.8/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado, automático]
libpcl-common1.8/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado, automático]
libpcl-conversions-dev/bionic,now 0.2.1-2 all [instalado]
libpcl-dev/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado]
libpcl-features1.8/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado, automático]
libpcl-filters1.8/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado, automático]
libpcl-io1.8/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado, automático]
libpcl-kdtree1.8/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado, automático]
libpcl-keypoints1.8/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado, automático]
libpcl-ml1.8/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado, automático]
libpcl-msgs-dev/bionic,now 0.2.0-6 all [instalado, automático]
libpcl-octree1.8/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado, automático]
libpcl-outofcore1.8/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado, automático]
libpcl-people1.8/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado, automático]
libpcl-recognition1.8/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado, automático]
libpcl-registration1.8/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado, automático]
libpcl-sample-consensus1.8/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado, automático]
libpcl-search1.8/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado, automático]
libpcl-segmentation1.8/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado, automático]
libpcl-stereo1.8/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado, automático]
libpcl-surface1.8/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado, automático]
libpcl-tracking1.8/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado, automático]
libpcl-visualization1.8/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado, automático]
pcl-tools/bionic-updates,now 1.8.1+dfsg1-2ubuntu2.18.04.1 arm64 [instalado]
ros-melodic-pcl-conversions/bionic,now 1.7.4-1bionic.20221025.224046 arm64 [instalado]
ros-melodic-pcl-msgs/bionic,now 0.2.0-0bionic.20221025.222605 arm64 [instalado]
ros-melodic-pcl-ros/bionic,now 1.7.4-1bionic.20221025.224548 arm64 [instalado]
ros-melodic-pcl-ros-dbgsym/bionic,now 1.7.4-1bionic.20221025.224548 arm64 [instalado]
ros-melodic-perception-pcl/bionic,now 1.7.4-1bionic.20221026.001313 arm64 [instalado, automático]
```
The problem is that I have made a code that uses the octree, to detect near neighbours based on the points and their depth, to see if it is an object or not and detect it as an obstacle.

The code is a simple callback, here it is:
```cpp
void cameraCallback(const sensor_msgs::PointCloud2::ConstPtr& input_cloud) {

// Convert PointCloud2 message to PointCloud<PointT> object
    pcl::PointCloud<pcl::PointXYZ>::Ptr cloud(new pcl::PointCloud<pcl::PointXYZ>);
    pcl::fromROSMsg(*input_cloud, *cloud);

    // Create OctreePointCloudSearch object with desired resolution
    // More resolution, better performance
    pcl::octree::OctreePointCloudSearch<pcl::PointXYZ> octree(0.01);
    octree.setInputCloud(cloud);
    octree.addPointsFromInputCloud();

    // Iterate over points and find neighboring points within a certain radius
    pcl::IndicesPtr indices(new std::vector<int>);
    std::vector<int> pointIdxRadiusSearch;
    std::vector<float> pointRadiusSquaredDistance;
    float radius = 0.1; // desired radius
    int min_neighbors = 5; // desired minimum number of neighbors
    int max_nn = 5;
    //float min_depth = 0.1;
    //float max_depth = 1.5;
    //pcl::octree::OctreeDepthRestriction<pcl::PointXYZ>::Ptr depth_constraint(new pcl::octree::OctreeDepthRestriction<pcl::PointXYZ>(min_depth, max_depth));
    pcl::PointCloud<pcl::PointXYZ>::Ptr group(new pcl::PointCloud<pcl::PointXYZ>);
    for (int i = 0; i < cloud->points.size(); i++)
[... 29 lines cut ...]
CMakeFiles/lidar_process.dir/src/lidar_process.cpp.o: In function `void pcl::detail::FieldMapper<pcl::PointXYZ>::operator()<pcl::fields::x>()':
lidar_process.cpp:(.text._ZN3pcl6detail11FieldMapperINS_8PointXYZEEclINS_6fields1xEEEvv[_ZN3pcl6detail11FieldMapperINS_8PointXYZEEclINS_6fields1xEEEvv]+0x1b0): undefined reference to `pcl::console::print(cl::console::VERBOSITY_LEVEL, char const*, ...)'
CMakeFiles/lidar_process.dir/src/lidar_process.cpp.o: In function `void pcl::detail::FieldMapper<pcl::PointXYZ>::operator()<pcl::fields::y>()':
lidar_process.cpp:(.text._ZN3pcl6detail11FieldMapperINS_8PointXYZEEclINS_6fields1yEEEvv[_ZN3pcl6detail11FieldMapperINS_8PointXYZEEclINS_6fields1yEEEvv]+0x1b4): undefined reference to `pcl::console::print(cl::console::VERBOSITY_LEVEL, char const*, ...)'
CMakeFiles/lidar_process.dir/src/lidar_process.cpp.o: In function `void pcl::detail::FieldMapper<pcl::PointXYZ>::operator()<pcl::fields::z>()':
lidar_process.cpp:(.text._ZN3pcl6detail11FieldMapperINS_8PointXYZEEclINS_6fields1zEEEvv[_ZN3pcl6detail11FieldMapperINS_8PointXYZEEclINS_6fields1zEEEvv]+0x1b4): undefined reference to `pcl::console::print(cl::console::VERBOSITY_LEVEL, char const*, ...)'
CMakeFiles/lidar_process.dir/src/lidar_process.cpp.o:(.data.rel.ro._ZTVN3pcl6octree22OctreePointCloudSearchINS_8PointXYZENS0_27OctreeContainerPointIndicesENS0_20OctreeContainerEmptyEEE[_ZTVN3pcl6octree22OtreePointCloudSearchINS_8PointXYZENS0_27OctreeContainerPointIndicesENS0_20OctreeContainerEmptyEEE]+0x30): undefined reference to `pcl::octree::OctreePointCloud<pcl::PointXYZ, pcl::octree::OctreeContainerPintIndices, pcl::octree::OctreeContainerEmpty, pcl::octree::OctreeBase<pcl::octree::OctreeContainerPointIndices, pcl::octree::OctreeContainerEmpty> >::addPointIdx(int)'
CMakeFiles/lidar_process.dir/src/lidar_process.cpp.o:(.data.rel.ro._ZTVN3pcl6octree22OctreePointCloudSearchINS_8PointXYZENS0_27OctreeContainerPointIndicesENS0_20OctreeContainerEmptyEEE[_ZTVN3pcl6octree22OtreePointCloudSearchINS_8PointXYZENS0_27OctreeContainerPointIndicesENS0_20OctreeContainerEmptyEEE]+0x38): undefined reference to `pcl::octree::OctreePointCloud<pcl::PointXYZ, pcl::octree::OctreeContainerPintIndices, pcl::octree::OctreeContainerEmpty, pcl::octree::OctreeBase<pcl::octree::OctreeContainerPointIndices, pcl::octree::OctreeContainerEmpty> >::genOctreeKeyForDataT(int const&, pcl::octree::OctreeKy&) const'
collect2: error: ld returned 1 exit status
hybrid_version/CMakeFiles/lidar_process.dir/build.make:115: recipe for target '/home/cartteclab/combined/devel/lib/hybrid_version/lidar_process' failed
make[2]: *** [/home/cartteclab/combined/devel/lib/hybrid_version/lidar_process] Error 1
CMakeFiles/Makefile2:2411: recipe for target 'hybrid_version/CMakeFiles/lidar_process.dir/all' failed
make[1]: *** [hybrid_version/CMakeFiles/lidar_process.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  3%] Built target _mqtt_ros_bridge_generate_messages_check_deps_Vector3Time
[  3%] Linking CXX executable /home/cartteclab/combined/devel/lib/hybrid_version/following
[  3%] Built target following
[  3%] Linking CXX executable /home/cartteclab/combined/devel/lib/hybrid_version/hybrid
[  3%] Built target hybrid
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```

The truth that I am something new with ROS and more using PCL, I think it is an error of the libraries or something like that, but I do not know what to do more digging in because this error comes out and I do not give with a solution that brings some light on the matter.

mvieth, 2023-03-22:
@AnOrdinaryUsser  Please post your CMakeLists.txt

AnOrdinaryUsser, 2023-03-22:
> @AnOrdinaryUsser Please post your CMakeLists.txt

```
# toplevel CMakeLists.txt for a catkin workspace
# catkin/cmake/toplevel.cmake

cmake_minimum_required(VERSION 3.0.2)

project(Project)

set(CATKIN_TOPLEVEL TRUE)

# search for catkin within the workspace
set(_cmd "catkin_find_pkg" "catkin" "${CMAKE_SOURCE_DIR}")
execute_process(COMMAND ${_cmd}
  RESULT_VARIABLE _res
  OUTPUT_VARIABLE _out
  ERROR_VARIABLE _err
  OUTPUT_STRIP_TRAILING_WHITESPACE
  ERROR_STRIP_TRAILING_WHITESPACE
)
if(NOT _res EQUAL 0 AND NOT _res EQUAL 2)
  # searching fot catkin resulted in an error
  string(REPLACE ";" " " _cmd_str "${_cmd}")
  message(FATAL_ERROR "Search for 'catkin' in workspace failed (${_cmd_str}): ${_err}")
endif()

# include catkin from workspace or via find_package()
if(_res EQUAL 0)
  set(catkin_EXTRAS_DIR "${CMAKE_SOURCE_DIR}/${_out}/cmake")
  # include all.cmake without add_subdirectory to let it operate in same scope
  include(${catkin_EXTRAS_DIR}/all.cmake NO_POLICY_SCOPE)
  add_subdirectory("${_out}")

else()
  # use either CMAKE_PREFIX_PATH explicitly passed to CMake as a command line argument
  # or CMAKE_PREFIX_PATH from the environment
  if(NOT DEFINED CMAKE_PREFIX_PATH)
    if(NOT "$ENV{CMAKE_PREFIX_PATH}" STREQUAL "")
      if(NOT WIN32)
        string(REPLACE ":" ";" CMAKE_PREFIX_PATH $ENV{CMAKE_PREFIX_PATH})
      else()
        set(CMAKE_PREFIX_PATH $ENV{CMAKE_PREFIX_PATH})
      endif()
    endif()
  endif()

  # list of catkin workspaces
  set(catkin_search_path "")
  foreach(path ${CMAKE_PREFIX_PATH})
    if(EXISTS "${path}/.catkin")
      list(FIND catkin_search_path ${path} _index)
      if(_index EQUAL -1)
        list(APPEND catkin_search_path ${path})
      endif()
    endif()
  endforeach()

  # search for catkin in all workspaces
  set(CATKIN_TOPLEVEL_FIND_PACKAGE TRUE)
  find_package(catkin QUIET
    NO_POLICY_SCOPE
    PATHS ${catkin_search_path}
    NO_DEFAULT_PATH NO_CMAKE_FIND_ROOT_PATH)
find_package(PCL REQUIRED)
include_directories(${PCL_INCLUDE_LIBRARIES})
  unset(CATKIN_TOPLEVEL_FIND_PACKAGE)

  if(NOT catkin_FOUND)
    message(FATAL_ERROR "find_package(catkin) failed. catkin was neither found in the workspace nor in the CMAKE_PREFIX_PATH. One reason may be that no ROS setup.sh was sourced before.")
  endif()
endif()

catkin_workspace()
```

mvieth, 2023-03-22:
@AnOrdinaryUsser Where do you link to the PCL libraries? (`target_link_libraries`)

Read this if you haven't already: https://pcl.readthedocs.io/projects/tutorials/en/master/using_pcl_pcl_config.html