ros-perception/perception_pcl #420: Failure to build ros2_ouster driver with perception_pcl source in workspace

sgvandijk, 2023-06-16:
I am trying to build both `pcl_conversions` and `ros2_ouster` from source. Mainly because  I work with Humble, and the following fix has not made it into the Humble release yet:
https://github.com/ros-perception/perception_pcl/pull/384
meaning that installing `ros-humble-pcl-conversions` pulls in 1.2GB of stuff. For the same reason I also do the same for `ros2_ouster`.

However, the build of `ros2_ouster` fails with the following error:
```
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/jsoncpp/jsoncpp-namespaced-targets.cmake:5 (add_library):
  add_library cannot create imported target "JsonCpp::JsonCpp" because
  another target with the same name already exists.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/jsoncpp/jsoncppConfig.cmake:41 (include)
  CMakeLists.txt:24 (find_package)
```

I am submitting this as an issue here rather than at `ros2_ouster`, because reverting https://github.com/ros-perception/perception_pcl/pull/392 results in a successful build. I.e., changing:
```
ament_export_dependencies(${dependencies} PCL)
```
back to
```
ament_export_dependencies(${dependencies})
```

I am not 100% exactly what happens, but I assume that `PCL` exports the `JsonCpp::JsonCpp` target, which conflicts with the target being created by `find_package(jsonccp REQUIRED)` in [`ros2_ouster`'s `CMakeLists.txt`](https://github.com/ros-drivers/ros2_ouster_drivers/blob/ros2/ros2_ouster/CMakeLists.txt). I think the way `ros2_ouster` finds and links to `jsoncpp` is reasonable/correct, so I feel that the issue then lies with `pcl_conversions` or PCL itself. But happy to file this somewhere else if that seems more appropriate.

## To reproduce:
With a ROS2 Humble installation:
```bash
mkdir testws
cd testws
mkdir src
git clone -b ros2 https://github.com/ros-perception/perception_pcl.git src/perception_pcl
git clone -b humble https://github.com/ros-drivers/ros2_ouster_drivers.git src/ros2_ouster_drivers
rosdep install --from-paths src/ -y -r --ignore-src
colcon build
```

closed 2024-04-04 by SteveMacenski

CharlesAntoineParent, 2024-04-15:
@sgvandijk Did you solve your problem ? I encountered the same problem and I am still trying to fix it !

skasperski, 2024-08-09:
I had the same issue and it seems to be caused by some strange behavior between PCL and jsoncpp.
PCL requires VTK, which in turn finds jsoncpp, but not by using jsoncpp's config.cmake, but by use of a custom Find-Macro.
 This then causes the conflict when jsoncpp is find_package'd later, because this checks for it's internal targets, but not for the namespaced  JsonCpp::JsonCpp, which it then just tries to create unconditionally.

As a workaround, I just do `find_package(jsoncpp)` (or whatever required jsoncpp)  after `find_package(pcl_conversions)`, which solved the problem for me. But this is really just a workaround and might not work in all situations.