ros-perception/perception_pcl #423: pcl-ros include error under ros2 humble enviornment

lightinfection, 2023-07-06:
Hi, I install ros-humble-pcl-ros and am trying to include the "point_cloud.hpp" header file for developing a project, but here's an error about the include file in it. I find the ros.h is sourced on the 45th row, but this should not be supported in ros2. May I know if it's a bug or if I misunderstand anything here? Thank you.

#include <ros/ros.h>
#include <pcl/point_cloud.h>
#include <pcl/point_traits.h>
#include <pcl/for_each_type.h>
#include <pcl/conversions.h>

mvieth, 2023-07-06:
Fixed on branch `ros2` (https://github.com/ros-perception/perception_pcl/commit/3bb07c07c84e67df25f7001ea7022b0b40149059)

felixf4xu, 2023-08-03:
@wkentaro I see your name on the maintainer list, could you help to close this issue?

this repo has too many _open_ issues but some of them could be closed.

jonselling, 2024-05-10:
These changes were made on the `ros2` branch, is it possible to backport these changes to humble so the apt packages get updated?

jonselling, 2024-08-01:
@SteveMacenski Pinging since you are a maintainer in pcl_ros as well, is there a way we can backport this fix to `humble`? It looks like it was this PR: https://github.com/ros-perception/perception_pcl/pull/410

andrewjong, 2024-08-14:
Bump, just tried to install ros2-humble-pcl-ros and it errored with ROS1. Can we update it in the Ubuntu package repository please? Thanks!

SteveMacenski, 2024-08-15:
Merged, will do release this afternoon :-)

andrewjong, 2024-08-15:
Hurray! Thank you so much!

SteveMacenski, 2024-08-15:
https://github.com/ros/rosdistro/pull/42460

closed 2024-08-15 by SteveMacenski

AbhiKhoyani, 2025-04-29:
I see this issue is closed that was for including ROS1 header. But in source code of `pcl_ros/include/point_cloud.hpp` there are still some references to ROS1. Such as [`Sensor_msgs::PointCloud2`](https://github.com/ros-perception/perception_pcl/blob/humble/pcl_ros/include/pcl_ros/point_cloud.hpp#L154) instead of `Sensor_msgs::msg::PointCloud2` and `ros::Time` and lot more. I've a lot other error as well like `point_cloud.hpp:152:8: error: ‘MD5Sum’ is not a class template
  152 | struct MD5Sum<pcl::PointCloud<T>>` Am I missing anything for building pcl-ros on humble? I'm using `humble` branch on docker ROS2 Humble.

Rayman, 2025-04-30:
Not every feature of perception_pcl is ported to ROS2, lots of features are still commented out.

It does build on humble I think, but only on the humble branch. I've tried to fix the CI here: https://github.com/ros/rosdistro/pull/45568