ros-perception/perception_pcl #424: No executables for "bag_to_pcd"

zillur-av, 2023-07-06:
I installed the package from source. I see it does not install the `bag_to_pcd` code. 
```
trc@rahzar:~$ ros2 pkg executables pcl_ros
pcl_ros filter_crop_box_node
pcl_ros filter_extract_indices_node
pcl_ros filter_passthrough_node
pcl_ros filter_project_inliers_node
pcl_ros filter_radius_outlier_removal_node
pcl_ros filter_statistical_outlier_removal_node
pcl_ros filter_voxel_grid_node
pcl_ros pcd_to_pointcloud
```
How do I get that?
enviroment: ros2 humble with ubuntu 22.04

felixf4xu, 2023-08-03:
because it's commented out https://github.com/ros-perception/perception_pcl/blob/ros2/pcl_ros/CMakeLists.txt#L168

    #add_executable(bag_to_pcd tools/bag_to_pcd.cpp)
    #target_link_libraries(bag_to_pcd pcl_ros_tf ${Boost_LIBRARIES} ${catkin_LIBRARIES} ${PCL_LIBRARIES})

schreiterjp, 2023-09-08:
Hi, commented in both lines as well as this [Line 223](https://github.com/ros-perception/perception_pcl/blob/e8ce25267a6d9d0bb0d80ba17e9c65269a7254aa/pcl_ros/CMakeLists.txt#L223) but there is no executable after running colcon build. Any suggestions?

themightyoarfish, 2024-12-23:
#301 removed this and many other exes

mikeferguson, 2024-12-23:
@themightyoarfish - it's not really "removed" - it just hasn't been ported to ROS 2. The API for rosbag is entirely different in ROS 1 and ROS 2 and so far it's not an effort anyone has put in

SelinnaL, 2025-01-30:
I have this problem on ROS2 humble distribution.

TomasCorral, 2025-06-26:
Same here, ROS2 humble

ros2 pkg executables pcl_ros
pcl_ros pcd_to_pointcloud

ros2 component types
pcl_ros
  pcl_ros::PCDPublisher