ros-perception/perception_pcl #432: `pcl::concatenatePointCloud` output row_step is incorrect

VRichardJP, 2023-11-16:
I observed on humble release that the output row_step of `concatenatePointCloud` is the row_step on the first input cloud.

It seems the behavior has been modified by:
https://github.com/ros-perception/perception_pcl/commit/33dbb3bf1fda7b7e5a3382cf36f8961b15d2d8b9

Now the output row_step is the output width, however I think row_step is normally equal to width*point_size.

mvieth, 2023-11-16:
Yes, you are right. `row_step` is the length of a row in bytes, so `cloud_out.row_step = cloud_out.width`  is not correct. `cloud_out.row_step = cloud_out.width * cloud_out.point_step` should do the right thing.

SteveMacenski, 2023-11-16:
Can you open a PR?

mvieth, 2023-11-16:
https://github.com/ros-perception/perception_pcl/pull/433

closed 2023-11-16 by SteveMacenski