ros-perception/perception_pcl #435: pcl_conversions incompatible with latest version of PCL

kgreenek, 2023-12-08:
When using the latest version of pcl with pcl_conversions, I see the following compiler errors:

```
pcl_conversions/include/pcl_conversions/pcl_conversions.h: In function 'void pcl_conversions::fromPCL(const pcl::PointIndices&, pcl_msgs::msg::PointIndices&)':
external/perception_pcl/pcl_conversions/include/pcl_conversions/pcl_conversions.h:304:25: error: no match for 'operator=' (operand types are 'pcl_msgs::msg::PointIndices_<std::allocator<void> >::_indices_type' {aka 'std::vector<int>'} and 'const Indices' {aka 'const std::vector<unsigned int, std::allocator<unsigned int> >'})
  304 |     pi.indices = pcl_pi.indices;
```

It appears as if the pcl::PointIndices type is an unsigned int, but the pcl_conversions code is expecting it to be a signed int.

SteveMacenski, 2023-12-08:
The main branch of this targets ROS 2 rolling, so if the rolling version of PCL doesn't have that change then we should keep it the way it is for compatability with the REP 2000 officially supported versions for a distribution.

However, we should definitely have this fixed for newer versions. I'd appreciate a PR to implement something like the following for the new version of PCL that this applies to:
```
#if PCL_VERSION_COMPARE(>=, Z, X, Y)
    // new setting
#else
    // old setting
#endif
```

mvieth, 2023-12-09:
@kgreenek Do you build pcl yourself, with `PCL_INDEX_SIGNED=false` ( https://github.com/PointCloudLibrary/pcl/blob/master/cmake/pcl_options.cmake#L99 ) ? If `PCL_INDEX_SIGNED=true` (the default), then `indices` in `pcl::PointIndices` is a vector of (signed) int. If `PCL_INDEX_SIGNED=false`, then `indices` in `pcl::PointIndices` is a vector of _unsigned_ int. Note: this is not a very recent change, it has been around for several PCL versions.
pcl_conversions currently indeed only works with the default case, `PCL_INDEX_SIGNED=true`.

kgreenek, 2023-12-16:
Oh! This is almost certainly my problem. I am using the bazel implementation and I just noticed that it has that flag set to false for some reason. I'll fix it in the bazel rules for building pcl. Thank you!

closed 2023-12-16 by kgreenek

SteveMacenski, 2023-12-16:
Happy to help! (Though @mvieth did all the thing-knowing)