ros-perception/perception_pcl #445: ros2 humble  pointcloud_to_pcd stuck

mgiurato, 2024-04-11:
Hi folks,
I am running in a ROS2 Humble environment a RTABmap node which provides me a topic /rtabmap/cloud_map (sensor_msgs/msg/PointCloud2) and I want it to be converted in a .pcd with the executable pointcloud_to_pcd.
From the terminal I am running the command:

`ros2 run pcl_ros pointcloud_to_pcd --ros-args --remap input:=/rtabmap/cloud_map`

After the execution I am not getting any feedback from the command line nor the process finishes the execution and I must kill it with ctrl+c.

Do you have any idea?
In particular, I am not running the pcl_ros package from apt but I succesfully built the code from the commit https://github.com/ros-perception/perception_pcl/commit/7e3401c96a9f58861693917f479f43ff3521f3c1 of the ros2 branch.

Have a nice day!

Mattia

samuel-ongzx, 2024-05-09:
Hi Mattia, 

I tried the `pointcloud_to_pcd` from commit [`7e3401c`](https://github.com/ros-perception/perception_pcl/commit/7e3401c96a9f58861693917f479f43ff3521f3c1) and I am able to save the PCD files from the point cloud topic. See image attached. 
[image]

The `pointcloud_to_pcd` node (being a node) will still be alive after saving all the point clouds (assuming you are replaying from a bag file). You would have to kill the node if you would like to stop the node from continuing to listen to the specific `PointCloud2` topic and saving the `pcd` files. 

Let me know if you still face this problem or if I misunderstood your answer, I would be happy to help.

CCodie, 2024-10-04:
Hi @samuel-ongzx ,
thank you for sharing information !

Now I can save pcd files from pointcloud2 topic,
but the problem is that points only have 'x, y, z' attributes when I save it into pcd file.

I want to use 'intensity, time' information but it seems they are lost during converting.
Can you give me some advice ?

KacperSynator, 2024-12-04:
Hi @CCodie,

I have similar issue. You can try changing `PoinXYZ` in [pointcloud_to_pcd.cpp](https://github.com/ros-perception/perception_pcl/blob/c1ccdd2776043ba1477926acceab7df75f64bf22/pcl_ros/tools/pointcloud_to_pcd.cpp#L135) to `PointXYZI`. 

For me it is still not working, i have the problem that is described here: https://github.com/lgsvl/simulator/issues/51#issuecomment-769550009