ros-perception/perception_pcl #448: Package Broken on Noble

AlexDayCRL, 2024-04-24:
This package is currently broken in the ROS2 repo as noble removed `libpcl-common1.13` in favor of `libpcl-common1.14`:

```
Package: ros-rolling-pcl-conversions
Version: 2.6.1-2noble.20240416.222509
Architecture: amd64
Maintainer: Paul Bovbel <paul@bovbel.com>
Installed-Size: 79
Depends: libeigen3-dev, libpcl-common1.13, libpcl-io1.13, ros-rolling-message-filters, ros-rolling-pcl-msgs, ros-rolling-rclcpp, ros-rolling-sensor-msgs, ros-rolling-std-msgs, ros-rolling-ros-workspace
Homepage: http://wiki.ros.org/pcl_conversions
Priority: optional
Section: misc
Filename: pool/main/r/ros-rolling-pcl-conversions/ros-rolling-pcl-conversions_2.6.1-2noble.20240416.222509_amd64.deb
Size: 12508
SHA256: 007a59c5c024bcf4db3e71d94556db984c3f5c7d07bcf437a2a9d930347f0432
SHA1: 8be154e27801a9a118255a9a17939d091c06cdba
MD5sum: 9731a62d6fc6d2afcb1f33cd9329963a
Description: Provides conversions from PCL data types and ROS message types
```

https://answers.launchpad.net/ubuntu/noble/amd64/libpcl-common1.13

SteveMacenski, 2024-04-24:
I can't seem to find where 1.13 is specified for pcl_conversions, do you see something? This may also be a transient issue, the migration to Rolling on 24.04 is still underway I believe so catching up with PCL updates for the distribution might not be completed yet. 

Or, it seems to me that perhaps rosdep just needs the update from 1.13 to 1.14 for the new distribution https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L4813 which should be easy enough to do in a contributed PR :smile:

AlexDayCRL, 2024-04-24:
Packaging is something that is still a mystery to me so I'm not sure where it's pulling it from. Interestingly some of the other ros-rolling packages specify libpcl-common1.13 and above rather than 1.13 only.

SteveMacenski, 2024-04-24:
Can you point to that? I can glance

AlexDayCRL, 2024-04-24:
I can't find any information as to why the ubuntu devs deleted 1.13 and went to 1.14. It seems to still be experimental in the debian upstream. If ROS is requiring PCL to 1.13 I suppose the onus is on them to provide the package rather than the devs to link against new versions. 

Would the rosdep repo be the best place to make an issue?

Here's where I'm finding the >= 1.13 dependency: 

http://packages.ros.org/ros2/ubuntu/dists/noble/main/binary-amd64/Packages

Cartographer is one such example:

```
Package: ros-rolling-cartographer-ros
Version: 2.0.9003-1noble.20240417.214619
Architecture: amd64
Maintainer: Chris Lalancette <clalancette@openrobotics.org>
Installed-Size: 26131
Depends: libabsl20220623t64 (>= 0~20220623.0-1), libboost-iostreams1.83.0 (>= 1.83.0), libc6 (>= 2.38), libcairo2 (>= 1.6.0), libceres4t64 (>= 2.2.0+dfsg), libgcc-s1 (>= 3.3.1), libgflags2.2 (>= 2.2.2), libgoogle-glog0v6t64 (>= 0.6.0), liblua5.2-0 (>= 5.2.4), libpcl-common1.13 (>= 1.13.0+dfsg), libprotobuf32t64 (>= 3.21.12), libstdc++6 (>= 13.1), libabsl-dev, libeigen3-dev, libgflags-dev, libgoogle-glog-dev, libpcl-dev, ros-rolling-builtin-interfaces, ros-rolling-cartographer, ros-rolling-cartographer-ros-msgs, ros-rolling-geometry-msgs, ros-rolling-launch, ros-rolling-nav-msgs, ros-rolling-pcl-conversions, ros-rolling-rclcpp, ros-rolling-robot-state-publisher, ros-rolling-rosbag2-cpp, ros-rolling-rosbag2-storage, ros-rolling-sensor-msgs, ros-rolling-std-msgs, ros-rolling-tf2, ros-rolling-tf2-eigen, ros-rolling-tf2-msgs, ros-rolling-tf2-ros, ros-rolling-urdf, ros-rolling-visualization-msgs, ros-rolling-ros-workspace
Homepage: https://github.com/cartographer-project/cartographer_ros
Priority: optional
Section: misc
Filename: pool/main/r/ros-rolling-cartographer-ros/ros-rolling-cartographer-ros_2.0.9003-1noble.20240417.214619_amd64.deb
Size: 3111294
SHA256: 38f311b881d334c516fa5096c0fafa7e80d9058a76988b6fc8a145122875fb56
SHA1: 38858e3bd9095affa3fd3593bc5ee5a556b9c82b
MD5sum: 19cc8b2e9903b5a4766d3d1082170ccb
Description: Cartographer is a system that provides real-time simultaneous localization and mapping (SLAM) in 2D and 3D across multiple platforms and sensor configurations.
 This package provides Cartographer's ROS integration.
```

SteveMacenski, 2024-04-25:
It looks like 1.14 is the right answer and the REP 2000 just needs to be updated. I think that should fix things once the PR linked above gets trickled out

clalancette, 2024-04-25:
So I *believe* this should all be settled now with the update to https://github.com/ros/rosdistro/pull/40834 and the re-bloom of `perception_pcl` that I did this morning.  At least, as far as I can tell there are no failing builds anymore.

closed 2024-04-25 by SteveMacenski