ros-perception/perception_pcl #458: ABI break on rolling and rosdistro issue

ahcorde, 2024-06-19:
We are planning to break ABI on message_filter I created this PR https://github.com/ros-perception/perception_pcl/pull/457 to fix the build job but I can see an issue in rosdistro.

There are branches for `humble`, `iron` and `rolling (ros2 branch)` but all rosdistro distributions are pointing to the `ros2` branch 

 - rolling https://github.com/ros-perception/perception_pcl/commits/humble/
 - jazzy https://github.com/ros/rosdistro/blob/master/jazzy/distribution.yaml#L4417-L4421
 - iron https://github.com/ros/rosdistro/blob/master/iron/distribution.yaml#L4381-L4385
 - humble https://github.com/ros/rosdistro/blob/master/humble/distribution.yaml#L5637-L5641

But I also check that there are some activity at least on the `humble` branch.

I would love to hear from the maintainers what to do here.
 - override `humble`, `iron` and `jazzy` with the actual `ros2` branch? and update rosdistro ? but this might destroy some work done for these old distros
 - create a new branch for humble iron and jazzy based on `ros2` and then continue working on ros2
 - any other ideas ?

SteveMacenski, 2024-06-19:
> but all rosdistro distributions are pointing to the ros2 branch

That's just the doc link. But, yeah, that's one of the downsides of rolling's auto-release into new distributions, it just fills in the rolling branch's fields that might not be appropriate and are easy to not catch when running releases since its non-software. When I run releases, its on the proper distribution branch. 

I don't think for a package like this any ABI breaking change is acceptable for anything as critical as message filters, that should be only done for Jazzy and newer. I hope you're not saying that `message_filters` is breaking for Humble and Iron, right? That would be **really bad** for users, I know alot of folks that have customizations on message filters and those ABI/API policies are in place for a reason. Obviously welcome to change between distributions though so `ros2` is game!

ahcorde, 2024-06-19:
New changes are only for `rolling`

SteveMacenski, 2024-06-20:
Oh yeah, then I think we're on board! Is the confusion just that the docs portion says `ros2` from the rolling auto-release? I think otherwise, throw that bad boy in `ros2` and we're good to go, humble/iron are on other branches

ahcorde, 2024-06-20:
I can see in rosdistro that `humble` is pointing to `ros2` and `iron` is also poiting to `ros2` branch (not only documentation)

 - https://github.com/ros/rosdistro/blob/master/iron/distribution.yaml#L4381-L4400
 - https://github.com/ros/rosdistro/blob/master/humble/distribution.yaml#L5651-L5656

SteveMacenski, 2024-06-20:
Check out the actual releases https://github.com/ros-perception/perception_pcl/releases, the software targets its branches.

Keep in mind that when rolling forks into a new distribution, it populates the rolling fields. It looks like I / we didn't `--edit` to update the auxiliary items like docs / source locations to the updated pose, but rest assured, the software itself is *all* pointed to their appropriate branches for the build farm and binaries.