ros-perception/perception_pcl #463: ros2 run pcl_ros bag_to_pcd: No executable found

juliaivonne, 2024-07-30:
I installed pcl_ros and uncommented the 2 lines in the CMake.txt-file:

```
add_executable(bag_to_pcd tools/bag_to_pcd.cpp)
target_link_libraries(bag_to_pcd pcl_ros_tf ${Boost_LIBRARIES} ${catkin_LIBRARIES} ${PCL_LIBRARIES})
```

But when I am trying to run it with ` $ rosrun pcl_ros bag_to_pcd <input_file.bag> <topic> <output_directory> ` I get "No executable found". Can anyone help me? I am using ROS2 Foxy on Ubuntu 20.04 and I am new to ROS in general.

mvieth, 2024-07-31:
> I installed pcl_ros

Via apt or did you build and install from source? If it is the latter, which commands did you use?

themightyoarfish, 2024-12-23:
Duplicate of https://github.com/ros-perception/perception_pcl/issues/424

for some reason, many executables were removed years ago, bag_to_pcd among them.