ros-perception/perception_pcl #475: dev dependencies listed as deps in released deb packages

grantweiss-RaymondCorp, 2025-01-14:
Our docker images were unusually large and I was able to track it down to the perception pcl packages. It looks like `libeigen3-dev, libpcl-dev` are listed as runtime dependencies in the control file. I switched to a source install of this package (
current jazzy branch c1ccdd2) since the `package.xml` looked correct and was able to remove 2gb+ in our image size. 

I tried digging into root cause with bloom and the repo at https://github.com/ros2-gbp/perception_pcl-release but am unfortunately not familiar with how it all works.

http://packages.ros.org/ros2/ubuntu/dists/noble/main/binary-amd64/Packages:
```
Package: ros-jazzy-pcl-ros
Version: 2.6.2-1noble.20241228.004807
Architecture: amd64
Maintainer: Paul Bovbel <paul@bovbel.com>
Installed-Size: 2717
Depends: libc6 (>= 2.34), libconsole-bridge1.0 (>= 1.0.1+dfsg2), libgcc-s1 (>= 3.3.1), libpcl-common1.14 (>= 1.14.0+dfsg), libpcl-filters1.14 (>= 1.14.0+dfsg), libpcl-io1.14 (>= 1.14.0+dfsg), libstdc++6 (>= 13.1), libeigen3-dev, libpcl-dev, libpcl-features1.14, libpcl-segmentation1.14, libpcl-surface1.14, ros-jazzy-geometry-msgs, ros-jazzy-pcl-conversions, ros-jazzy-rclcpp, ros-jazzy-rclcpp-components, ros-jazzy-sensor-msgs, ros-jazzy-tf2, ros-jazzy-tf2-eigen, ros-jazzy-tf2-geometry-msgs, ros-jazzy-tf2-ros, ros-jazzy-ros-workspace
Homepage: http://ros.org/wiki/perception_pcl
Priority: optional
Section: misc
Filename: pool/main/r/ros-jazzy-pcl-ros/ros-jazzy-pcl-ros_2.6.2-1noble.20241228.004807_amd64.deb
Size: 455774
SHA256: 2ac1742f033ed1d298631ab94785c28aa471af1e3d297f79f53126ebf46e0ae1
SHA1: 6d929256a56f2dc08a2e85c140bc0c22218d1578
MD5sum: 58b4b6ac5f0f8703d4d38e8f2df9ddd9
Description: PCL (Point Cloud Library) ROS interface stack.
 PCL-ROS is the preferred bridge for 3D applications involving n-D Point Clouds and 3D geometry processing in ROS.
```

```
Package: ros-jazzy-pcl-conversions
Version: 2.6.2-1noble.20241228.002523
Architecture: amd64
Maintainer: Paul Bovbel <paul@bovbel.com>
Installed-Size: 79
Depends: libeigen3-dev, libpcl-common1.14, libpcl-dev, libpcl-io1.14, ros-jazzy-message-filters, ros-jazzy-pcl-msgs, ros-jazzy-rclcpp, ros-jazzy-sensor-msgs, ros-jazzy-std-msgs, ros-jazzy-ros-workspace
Homepage: http://wiki.ros.org/pcl_conversions
Priority: optional
Section: misc
Filename: pool/main/r/ros-jazzy-pcl-conversions/ros-jazzy-pcl-conversions_2.6.2-1noble.20241228.002523_amd64.deb
Size: 12910
SHA256: ff5c99594081bd65a60e8134acc958d555e4edf6dcbec10171cf729c83a23f00
SHA1: 127d5cf2fbe7ac9becfb7491dab9249ab65dbc2a
MD5sum: f2908798356b4b7347c14625d78f63db
Description: Provides conversions from PCL data types and ROS message types
```

Rayman, 2025-01-15:
Hmm interesting, In theory we should not have a `exec_depend` on eigen and pcl-dev. But I don't know how to exactly change the `package.xml` to get this behavior.

SteveMacenski, 2025-01-15:
I don't see that pcl-ros has an exec depend: https://github.com/ros-perception/perception_pcl/blob/ros2/pcl_ros/package.xml#L45-L50

grantweiss-RaymondCorp, 2025-01-15:
That's what confused me. The package.xml all looked correct and builds fine from source. It seems the process of building this into a prebuilt package is the issue?

grantweiss-RaymondCorp, 2025-01-15:
This could be it? It looks like buildtool_export_depends are added as runtime dependencies in the debian package.

https://github.com/ros-infrastructure/bloom/blob/b42a1dd043d0712bca7e37dad91c4a54752024d8/bloom/generators/debian/generator.py#L349-L372

SteveMacenski, 2025-01-15:
I believe that `buildtool_export_depends` is required - that was added in https://github.com/ros-perception/perception_pcl/pull/447. But perhaps there's a different solution, but I will fully admit `buildtool_export_depends` is not a tag I have an intuition regarding. Maybe @Yadunund can offer some thoughts

Rayman, 2025-01-16:
Note that there is a difference between `build_export_depend` and `buildtool_export_depends`.

Rayman, 2025-01-16:
When I change to `package.xml` to only include this line `<build_export_depend>libpcl-all-dev</build_export_depend>`. And then run
```sh
bloom-generate rosdebian
```
It generates the following control file:
```
Source: ros-jazzy-pcl-conversions
Section: misc
Priority: optional
Maintainer: Paul Bovbel <paul@bovbel.com>
Build-Depends: debhelper (>= 9.0.0), ros-jazzy-ament-cmake
Homepage: http://wiki.ros.org/pcl_conversions
Standards-Version: 3.9.2

Package: ros-jazzy-pcl-conversions
Architecture: any
Depends: ${shlibs:Depends}, ${misc:Depends}, libpcl-dev
Description: Provides conversions from PCL data types and ROS message types
```
This is how the libpcl-dev ends up in the dependencies. But the `build_export_depend` is needed for other packages to compile against this package. So I don't know the solution.

grantweiss-RaymondCorp, 2025-01-16:
I think I found where it gets pulled in.

Bloom uses a `package.run_depends`, which is distinctly different from `package.exec_depends`. I ran a modified bloom with exec_depends instead and it did not pull in the `build_export_depend` dependencies (libpcl-dev) .
https://github.com/ros-infrastructure/bloom/blob/b42a1dd043d0712bca7e37dad91c4a54752024d8/bloom/generators/debian/generator.py#L350C25-L350C44

`package.run_depends` (not a native key in package.xml) are defined as `exec_depends` + `build_export_depends` in the catkin_pkg repo
https://github.com/ros-infrastructure/catkin_pkg/blob/eec17e72cbe3c25173ec0cc8868106904397c560/src/catkin_pkg/package.py#L118-L122

SteveMacenski, 2025-01-16:
Is there a change in bloom/catkin_pkg that should be made?

grantweiss-RaymondCorp, 2025-01-16:
I think this could be solved in bloom alone. It looks like catkin_pkg will backfill the `exec_depends` and `build_export_depends` keys on older packages using package version 1. So bloom could possibly just change to using `exec_depends` instead of `run_depends`. I would open a PR but I don't have the bandwidth for compatibility testing.

Yadunund, 2025-01-16:
> I think this could be solved in bloom alone. It looks like catkin_pkg will backfill the `exec_depends` and `build_export_depends` keys on older packages using package version 1. So bloom could [...]

I'm not the most familiar on the topic but starting a discussion in bloom/catkin_pkg would be best to get inputs from experts on the matter.

cottsay, 2025-01-16:
I responded to the Bloom ticket, but tl;dr is that we can't take advantage of the split between exec and build_export dependencies in deb packages until we split ROS packages into separate runtime and dev subpackages.

SteveMacenski, 2025-01-16:
That makes sense to me

Rayman, 2025-01-17:
These posts are very interesting to read. I think it could be solved by splitting packages into -dev packages, but I don't think this will be solved in the short term unfortunately:

https://discourse.ros.org/t/generating-dev-and-runtime-artefacts-from-ros-packages/12448
https://discourse.ros.org/t/package-separation-for-building-and-running-ros-applications/35527