ros-perception/perception_pcl #482: Getting build error while compiling pcl_ros after running colcon on ros2

kkarnatak, 2025-03-14:
$~/ros2_ws/src/pcl_ros/src/pcl_ros/filters/filter.cpp:187:57: error: ‘rclcpp::MatchedInfo’ has not been declared

  187 |   pub_options.event_callbacks.matched_callback = [this](rclcpp::MatchedInfo & /*info*/) {

Command executed:
$ colcon build

SteveMacenski, 2025-03-14:
Are you building the branch of the project that corresponds to your ROS 2 distribution?

AbhiKhoyani, 2025-04-28:
Yes, I also experience this error:

I'm using ROS2 humble and so using `-b humble` branch. Tried `-b ros2` branch as well but same error!

Rayman, 2025-04-29:
The `ros` and `jazzy` branch contain the word `rclcpp::MatchedInfo`. The humble branch doesn't contain this word. So this must mean you are using the wrong branch.

On humble, please checkout the humble branch.

AbhiKhoyani, 2025-04-29:
Thanks @Rayman I removed complete directory and cloned it again directly on humble branch with `git clone -b humble https://github.com/ros-perception/perception_pcl.git` and it did the trick! May be previously I might have made some mistake on checkout.

closed 2025-04-29 by SteveMacenski