ros-perception/perception_pcl #484: Dynamic reconfiguration server ROS 1 leftover

diogovazguerra, 2025-04-04:
Hi, I'm having issues building my lidar custom project that depends on some pcl_ros libraries. 

I'm using ubuntu 20.04 and I'm building my ROS 2 Humble workspace from source on it.

That problem is that some source codes from pcl_ros package still uses dynamic reconfiguration server that sets parameters from a cfg file, something that only a ROS 1 workspace could ever do. Even on the must recent version of the perception_pcl repository still has dynamic reconfiguration usage. 

I wanted to confirm if there is any other alternative to fix this issue or is there something else I'm missing.

Here's the compilation error output:

--- stderr: lidar
                               
In file included from ~/ros2_ws/src/perception/lidar/include/lidar/lidar_euclidian_clustering.h:4,
                 from ~/ros2_ws/src/perception/lidar/src/lidar_euclidian_clustering.cpp:1:
~/ros2_ws/install/pcl_ros/include/pcl_ros/pcl_ros/segmentation/extract_clusters.hpp:41:10: fatal error: dynamic_reconfigure/server.h: No such file or directory
   41 | #include <dynamic_reconfigure/server.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/euclidian_clustering.dir/build.make:63: CMakeFiles/euclidian_clustering.dir/src/lidar_euclidian_clustering.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:119: CMakeFiles/euclidian_clustering.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

In file included from ~/ros2_ws/src/perception/lidar/include/lidar/lidar_dbscan.h:4,
                 from ~/ros2_ws/src/perception/lidar/src/lidar_dbscan.cpp:1:
~/ros2_ws/install/pcl_ros/include/pcl_ros/pcl_ros/segmentation/extract_clusters.hpp:41:10: fatal error: dynamic_reconfigure/server.h: No such file or directory
   41 | #include <dynamic_reconfigure/server.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/dbscan_clustering.dir/build.make:63: CMakeFiles/dbscan_clustering.dir/src/lidar_dbscan.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:233: CMakeFiles/dbscan_clustering.dir/all] Error 2

In file included from ~/ros2_ws/src/perception/lidar/include/lidar/lidar_ransac.h:4,
                 from ~/ros2_ws/src/perception/lidar/src/lidar_ransac.cpp:1:
~/ros2_ws/install/pcl_ros/include/pcl_ros/pcl_ros/segmentation/sac_segmentation.hpp:43:10: fatal error: dynamic_reconfigure/server.h: No such file or directory
   43 | #include <dynamic_reconfigure/server.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ransac.dir/build.make:63: CMakeFiles/ransac.dir/src/lidar_ransac.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:92: CMakeFiles/ransac.dir/all] Error 2

In file included from ~/ros2_ws/src/perception/lidar/include/lidar/lidar_ransac.h:4,
                 from ~/ros2_ws/src/perception/lidar/include/lidar/lidar_pipeline.h:9,
                 from ~/ros2_ws/src/perception/lidar/src/lidar_visualization.cpp:1:
~/ros2_ws/install/pcl_ros/include/pcl_ros/pcl_ros/segmentation/sac_segmentation.hpp:43:10: fatal error: dynamic_reconfigure/server.h: No such file or directory
   43 | #include <dynamic_reconfigure/server.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/visualization.dir/build.make:63: CMakeFiles/visualization.dir/src/lidar_visualization.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:314: CMakeFiles/visualization.dir/all] Error 2

Rayman, 2025-04-07:
Some parts of pcl_ros are not yet ported to ros2. Any help or PRs would be welcome :)

mariamercade, 2025-06-13:
Hi, I'm also facing this issue trying to use sac_segmentation in ROS 2.

@Rayman, I'm considering contributing the migration myself (just for `sac_segmentation` for now) but before starting I’d like to ask a few things: Is anyone already working on migrating this part? Is there any particular reason why it hasn’t been migrated yet? Are there any known technical blockers or challenges?

Overall, is there anything important I should know before getting started?

diogovazguerra, 2025-07-30:
> Hi, I'm also facing this issue trying to use sac_segmentation in ROS 2.
>
> [@Rayman](https://github.com/Rayman), I'm considering contributing the migration myself (just for `sac_segmentation` for now) [...]

Hi @mariamercade , right now I also do want to contribute to the migration of sac_segmentation in ROS 2 starting today. If you already had some progress on the migrated package please let me know so I could help you accomplish that goal, if not that's okay I will make the migration process by myself.

Please be free to contact me through this issue.

diogovazguerra, 2025-07-30:
Also keep in mind I will be using the "humble" branch as "ros2" branch of this repository contains features that ros2 humble that I'm using for my project is not compatible with

diogovazguerra, 2025-07-30:
> > > Hi, I'm also facing this issue trying to use sac_segmentation in ROS 2.
> > > [@Rayman](https://github.com/Rayman), I'm considering contributing the migration myself (just for `sac_segmentation` for now) but before starting I’d like to ask a few things: Is anyone already working on migrating this part? Is there any particular reason why it hasn’t been migrated yet? Are there any known technical blockers or challenges?
> > > Overall, is there anything important I should know before getting started?
> > 
> > 
> > Hi [@mariamercade](https://github.com/mariamercade) , right now I also do want to contribute to the migration of sac_segmentation in ROS 2 starting today. If you already had some progress on the migrated package please let me know so I could help you accomplish that goal, if not that's okay I will make the migration process by myself.
> > Please be free to contact me through this issue.
> 
> In the end, I could manage to use directly pcl library so I did not start that migration. Go ahead!

@mariamercade are you using ros2 from source or from binary packages? Also what version of pcl library you are using and how did you managed to use on your ros2 workspace?

diogovazguerra, 2025-08-12:
Hi. I just wanted to tell that I just made a pull request of the various files migrated to ROS2 from pcl_ros package, but there was some github tests errors, must of it are lint errors. Wanted to know if any mantainer could help me know is that a real issue or not.